Duet3D/RepRapFirmware #376: M486 object cancellation information is not retained when resuming after fault.

christophmuellerorg, 2020-04-04:
I just played with the object cancelation feature and by chance had a heater error resulting in the print being paused. Once resumed the previously cancelled object print was continued and I was unable to cancel the same object a second time.

To reproduce: 
- start a print with multiple objects
- cancel at least one object
- have a heater failure automatically halt your print
- resume

Expected behavior:
- the object cancelation information will be retained for resuming
- object cancelation should be possible even after resuming a job

dc42, 2020-06-11:
Thanks for your excellent report. I've added this to the work list for RRF 3.2.

T3P3, 2023-05-04:
Fixed in RRF 3.2:

- The resurrect.g file now records which objects on the build plate have been cancelled

closed 2023-05-04 by T3P3