Duet3D/RepRapFirmware #381: Feature request: minimal fan speed for thermostatically controlled fans

khumarahn, 2020-04-18:
Hi. It would be nice to have a minimal speed for thermostatically controlled fans. A use case is for an exhaust fan from an enclosure, which should be always on because it filters the air, and which could increase speed if the air inside is getting too hot.

Billtopia, 2020-04-18:
It is already a thing... https://github.com/dc42/RepRapFirmware/issues/369

khumarahn, 2020-04-18:
That's minimal speed when a non-zero speed is requested... Say, I would like a fan which is at 20% when the sensor is below 50C, and changes from 20 to 100& between 50C and 60C. If I just set
```M106 L0.2 T50:60 ...```
then the fan will be at 0% below 50C.

Billtopia, 2020-04-18:
try setting the S to the speed you want when not in your temp range like S0.2 on the M106 define line

khumarahn, 2020-04-18:
I think I tried that, and it had no effect...

Billtopia, 2020-04-18:
I am printing at the moment so I can't try it, however you could space the temp range out further for now. like have the fan kick on at 30 or so at 10%

khumarahn, 2020-04-18:
Yes, I also thought of this. But I'd like a sharper increase around a specific temperature.
I just tried `M106 P1 S0.75 T70:80 H4` with H4 at 65C. The fan stopped.

Billtopia, 2020-04-18:
Try setting 'L' on the M106 line to something like L0.75 and see if that helps

khumarahn, 2020-04-18:
It  does not, also tried.

Billtopia, 2020-04-18:
Well... I tried and no joy... I suppose, if you are using a newer version of RRF that has the daemon.g thread you could use it and some meta commands to control the fan directly. It would take a bit of manual coding to do

khumarahn, 2020-04-18:
I use RRF 3.0 on Duet 2 wifi. What is daemon.g? Can you give me a pointer or an example program?

Billtopia, 2020-04-18:
it was announced here   https://github.com/dc42/RepRapFirmware/blob/v3-dev/WHATS_NEW_RRF3.md#reprapfirmware-301-rc3   it runs in the background and can do anything a normal gcode can do. so using the object model and https://duet3d.dozuki.com/Wiki/GCode_Meta_Commands it could control the fan with some if/elses

khumarahn, 2020-04-18:
Interesting, I will look at it. Thanks!

closed 2023-05-04 by T3P3