Duet3D/RepRapFirmware #384: Extruder motor chokes - speed limitation(?)

moo16, 2020-04-22:
hi,
i was calibrating values for my extruder motor when the motor starts to "choke" at certain speed.
i tried change the value all over but nothing release it from this speed limit.
im sure with most of my calibration data, 'step per mm' is probably not accurate but still im swinging between stall guard when the 'step per mm' is high and this choke(disturbing noise) when the 'step per mm' is lower.
i tried to add stall detection commands, i tried to work the extruder motor with no load(tourqe) on it but both actions with no help.
after the choking the motor is stuck like this until emergency stop is happened and not able to move.
im sure my motor can move faster his datasheet support it. 
is anybody happened the same? what the meaning of this noise? how is jerk sounds like?
happy for any help thank you  before head

moo16, 2020-04-22:
im using Duet 3 Mainboard 6HC
and Duet 3 Expansion 3HC
on web control

chrishamm, 2020-04-22:
Please use the [forum](https://forum.duet3d.com) for support requests.

closed 2020-04-22 by chrishamm