Duet3D/RepRapFirmware #385: RepRapFirmware 3.01-RC9: Thermostatically controlled fan does not turn on

Billtopia, 2020-04-22:
after upgrading to RepRapFirmware 3.01-RC9 from RC8 the thermostatically controlled fan does not turn on at the set temp. (As this is the hot end fan this caused a huge clog in the printer) The JSON string shows as thermostatically controlled by heater 1 with the proper temp set point but it doesn't turn on when the heater temperature passed the set point.

I rolled back to RC8 so I could print.

dc42, 2020-04-23:
Please use the forum for support. Start a new thread, posting your config.g file, board type and other relevant data.

Billtopia, 2020-04-23:
Well... it was more of a bug report than a support request

dc42, 2020-04-23:
Please don't jump to the conclusion that you have found a bug. You have provided far too little information for me to conclude that. Nobody else has reported this, and several of us including me have done tests prints on machines with thermostatic fans. There may be a bug that only manifests itself under particular conditions, or perhaps no bug at all but an error in your configuration.

Billtopia, 2020-04-25:
so Duet Wifi 2   RRFW 3.01-RC9 & RC10
I can tell you that it is a change in the handling of the M106 command.
If you include a 'S0' on the command setting up a thermostatically controlled fan  the  rr_model JSON returns
`{"actualValue":0,"blip":0.10,"max":1.00,"min":1.00,"name":"","requestedValue":0,"rpm":-1,"thermostatic":{"heaters":[1],"highTemperature":55.0,"lowTemperature":55.0}} `
and the fan doesn't engage at temp.
Without the 'S0' the rr_model JSON is (with the fan off)
`{"actualValue":0,"blip":0.10,"max":1.00,"min":1.00,"name":"","requestedValue":1.00,"rpm":-1,"thermostatic":{"heaters":[1],"highTemperature":55.0,"lowTemperature":55.0}}}`
and the fan does work
The S came from the config tool ( but seems to not be needed for thermostatically controlled fans)

dc42, 2020-04-25:
Thanks for the detail. The online configurator should not include S0 in the M106 line. Previous versions of RRF applied a minimum S0.5 to thermostatic fans, and it appears from your report that this is no longer the case.

Billtopia, 2020-04-26:
The config was generated in Jan. I just did a generic config and it is now doing "S1"

Billtopia, 2020-04-26:
The config was created in Jan. I just created a generic config and now it does "S1" My config must of gotten a hiccup.

closed 2020-10-20 by T3P3