Duet3D/RepRapFirmware #386: E0 filament sensor impacts E1 extrusion

oseiler2, 2020-04-23:
I'm using a Laser filament sensor on E0 with a filament switch. When there's no filament present in E0 I cannot print with another extruder (E1), but I get an error 'Extruder 0 reports no filament'

FW: 3.01-RC9 (2020-04-21b1)

oseiler2, 2020-04-23:
test.gcode
```
M591 D0 P6 C"e0_stop" S0 R20:200
G90
M83
T1
G1 X10 Y10 Z5
G1 X20 Y10 E0.1 F1500
```

eventlog:
```
2020-04-23 13:11:05 Started printing file 0:/gcodes/test.gcode
2020-04-23 13:11:05 Resume state saved
2020-04-23 13:11:05 Extruder 0 reports no filament
2020-04-23 13:11:05 Printing paused at X20.0 Y17.8 Z5.0
```

StephenRC, 2020-04-23:
Support is in https://forum.duet3d.com/

You could turn off the filament senser for E0.

On Wed, 22 Apr 2020 18:17:34 -0700, Oliver Seiler
<notifications@github.com> had a flock of green cheek conures squawk
out:

>I'm using a Laser filament sensor on E0 with a filament switch. When there's no filament present in E0 I cannot print with another extruder (E1), but I get an error 'Extruder 0 reports no filament'
>
>FW: 3.01-RC9 (2020-04-21b1)

T3P3, 2020-10-20:
please can you confirm if this issue is resolved in the latest release

oseiler2, 2020-10-21:
This still happens in 3.1.1 (2020-05-19b2) - I'll try 3.2b3 later tonight

oseiler2, 2020-10-21:
Confirming that this also still happens under RepRapFirmware for Duet 2 WiFi/Ethernet 3.2-beta2 (2020-10-05b2)

supertjj, 2021-03-25:
Still happens in RepRapFirmware for Duet 2 WiFi/Ethernet 3.2.2, I don't have any filament sensor for e0 and I have been using a passive pen tool while waiting for parts to build my hotend and extruder but once I wired up the hotend and extruder I started getting the same errors as above plus I am also getting this error "attempting to extrude with no tool selected" but in the gcode file I'm trying to run doesn't reference any E commands to extrude and it pauses and gives these errors at the very beginning of the code and the first line of the code is " T2" so how can it complain about extruding without a tool

supertjj, 2021-03-30:
My problem was created by config.g having a filament runout sensor declared for extruder 0 that's my fault, however the runout sensor for extruder 0 should not stop a print that uses another tool/extruder from running

T3P3, 2023-05-04:
Please open a forum discussion about this if its still an issue in RRF 3.4.5