Duet3D/RepRapFirmware #387: Console sorting in reverse date order  - not critical issue

carloperatoner, 2020-04-23:
I just noticed that the console sorting does not display properly in reverse date order when spanning days.   The previous day is shown first, then the current day  see below:

4/22/2020, 6:17:36 PM	G29
145 points probed, min error -0.113, max error 0.119, mean -0.008, deviation 0.042
Height map saved to file 0:/sys/heightmap.csv
4/22/2020, 6:12:54 PM	Calibrated 6 factors using 25 points, (mean, deviation) before (-0.002, 0.120) after (0.000, 0.023)
4/23/2020, 12:25:51 AM	M500
4/23/2020, 12:25:24 AM	G29
145 points probed, min error -0.137, max error 0.153, mean 0.000, deviation 0.048
Height map saved to file 0:/sys/heightmap.csv

chrishamm, 2020-04-23:
Provided you mean DWC (which has its own repository btw) you can click on the "Date" column to change the sorting order. I'll check the defaults again.

closed 2020-04-23 by chrishamm

chrishamm, 2020-04-23:
Just checked this again and the default sorting order is correct.

chrishamm, 2020-04-23:
PS: Just saw that the order of times/dates varies in your log. I'll try to reproduce that now.

chrishamm, 2020-04-23:
Please upgrade to the latest DWC version, I had fixed this problem many versions ago. It works as expected in the latest DWC 2.1.4:

[image]

carloperatoner, 2020-04-23:
Thank you,  I did not realize there was a separate download for DWC.     I just downloaded and will test tonight.   Thanks for all the great work.   The Duet Board and the firmwares have made my 3d printer a pleasure to use.

Carlo

[image]<https://user-images.githubusercontent.com/5919449/80092258-bea51300-8562-11ea-842e-a6ace8786989.png>