Duet3D/RepRapFirmware #389: 3.01-rc9 thermostatic fan at full speed

westminsterflip, 2020-04-24:
Upgraded from 3.0 to 3.01-rc9 and the thermostatic fan I had set up will only run full speed.  

It's set up with:
`M950 F2 C"fan2" Q50000`
`M106 P2 H2:3:4 B.1 T40:65`

When a sensor hits 40c the fan goes full tilt.

tekkydave, 2020-04-27:
Same here following upgrade from 3.01RC6 to 3.01RC9

I'm using the mcu-temp and drivers valurs to drive my enclosure fans on duex fan4 pins.
As soon as it goes over 26C the fans are on 100%

M308 S2 Y"mcu-temp" 
M308 S3 Y"drivers"

M950 F4 C"duex.fan4" Q500 
M106 P4 S1.0 H2:3 T26:65 L0.4 C"Enclosure Fans"

Billtopia, 2020-04-27:
Try removing the 'S1' and see

tekkydave, 2020-04-27:
Makes no difference. I've raised this on the Duet Forum and DC42 has accepted it is a bug and will be fixed in next RC release.

closed 2020-05-15 by westminsterflip

dc42, 2020-05-16:
Was fixed in 3.10-RC10.