Duet3D/RepRapFirmware #391: Wait for E0 endstop does not work with 3.01RCx, but works with 3.0

boa-pe, 2020-04-25:
I am using macro for loading filament
M577 E0 P"!e0stop" ; Wait for the filament sensor
I have filament presence sensor connected to E0 endstop inputs.

It looks like this wait never ends regardles of E0  input state

chrishamm, 2020-04-30:
This feature is working well, hence I'm closing this issue. You must not specify the pin in M577 but the level via S0 if you want to wait for the pin to go to low.

closed 2020-04-30 by chrishamm

boa-pe, 2020-05-10:
> 
> 
> This feature is working well, hence I'm closing this issue. You must not specify the pin in M577 but the level via S0 if you want to wait for the pin to go to low.

On which version You tested. I investigated that is stopped working on my board with 3.01RC2. I tried M577 E0 S0, but it sill does not work.

chrishamm, 2020-05-10:
3.01-RC12. Note that extruder endstops cannot be configured using M574 any more, so probably that is your problem. To achieve what you are trying to do you'll need something like

```
M950 J0 C"e0stop" ; configure general-purpose input pin 0 on e0stop
M577 P0 S0 ; wait for pin 0 to be low
```

Please use the [forum](https://forum.duet3d.com) for further support questions.