Duet3D/RepRapFirmware #394: M42 S parameter does not respond to values in 0-255 format.

jslaker, 2020-05-03:
The M42 command does not seem to be properly responding when passed an S parameter in the format of 0-255 in RRF3.01 RC11. Use case is in controlling brightness of LED light strips via PWM.

Example config:
```
M950 P0 C"!exp.heater3"		; Configure front LED strip
M950 P1 C"!exp.heater4"		; Configure side LED strip

M42 P0 S0.60 			; Set power-on brightness
M42 P1 S0.60			; 
```

Using S0.0-1.0 values behaves as expected. Replacing the S parameter with a 0-255 value results in no change in brightness on the LEDs. The Duet Gcode docs indicate that either should work and 0-255 were working with the same printer on RRF2.x.

I have talked to a couple of other users who saw similar behavior running their LEDs off Duex fan headers, so it does not appear to be specific to my configuration.

dc42, 2020-05-04:
I can't reproduce this. It's working fine for me on Duet Wii funning RRF 3.10-RC11.

jslaker, 2020-05-04:
That's very interesting. I'm unable to replicate it now myself as well. 

The one additional bit of context, and I'm not sure that it matters, is that I saw this behavior immediately after upgrading from RRF2 to RRF3 while updating my config.g. That was also the case for the person I was speaking to yesterday and what prompted the bug report. 

In both cases, the call was unresponsive (i.e., the LEDs didn't come on at all) when making M42 calls with S255 values. Changing it to call S1 immediately worked in both cases. I hadn't tried calling it again with 0-255 values until just now, but it does appear to be working. I'm not sure why that would be the case.

dc42, 2020-05-04:
Please re-open if you find a way to reproduce this.

closed 2020-05-04 by dc42