Duet3D/RepRapFirmware #398: [3.01RC12] M591 Filament runout sensor does not work when M950 is ued on E0stop input

boa-pe, 2020-05-11:
Duet2WiFi

In config.g
M950 J0 C"e0stop"
M591 D0 P1 C"e0stop" S1

M591 gives error:
Error: M591: Pin 'e0stop' is not free

And filament runout detection is not working anymore

chrishamm, 2020-05-11:
M591 must not be used with a GPIN pin / M950. Use
```
M591 D0 P1 C"e0stop" S1
```
instead.

As per your last issue (where I already asked you to post support questions on the [forum](https://forum.duet3d.com)), you'll have to change the M577 code to something like
```
while sensors.filamentSensors[0].filamentPresent != 0
  G4 P250
```
I will not respond to further support questions here.

closed 2020-05-11 by chrishamm

boa-pe, 2020-05-13:
Perhaps some consistency in pins naming/passing to commands would solve such problems. I could bet that making M591 to use pin number like M577 instead of "e0stop" would solve this issue.

Djlce, 2022-05-13:
M591 D0 P1 C3 S1                                   ; filament monitor connected to E0_stop

Have luck ;)