Duet3D/RepRapFirmware #400: G31 or M500 P31 not saving the current Z probe offset

ernstc, 2020-05-15:
My configuration is RepRapFirmware 3.01 RC12 + Duet3 + SBC

In config.g I have the command
```
G31 P500 X-22.65 Y7.35 Z1.61
```
so the Z probe trigger height is **1.61**.

I want to change the trigger height from the console to the value 1.53, so I execute the command
```
G31 Z1.53
```
and if I execute the command G31 without parameter I can see the canges:

```
G31
Z probe 0: current reading 1000, threshold 500, trigger height 1.530, offsets X-22.6 Y7.3

```
But if I refresh the browser, sniffing the network communications, the printer APIs return the first value: 1.61.
Also I want to save the settings running "**M500 P31**" command, but it saves the old value and not the new one:
```
; Z probe parameters
G31 P500 X-22.6 Y7.3 Z1.61

```

ernstc, 2020-05-15:
The issue is present also in the latest release: 
RepRapFirmware 3.1.0 + DSF 3.1.0

dc42, 2020-05-18:
Believed fixed in the forthcoming 3.1.1 release. Please test again when that release is available.

ernstc, 2020-05-20:
@dc42, I confirm that the issues has been resolved in the 3.1.1 release. Thanks!

closed 2020-05-20 by ernstc

ernstc, 2020-05-20:
Anyway the command **M500 P31** must be executed after almost 1 sec after the command **G31 Zxx** for saving the new value, otherwise still saves the old value.

chrishamm, 2020-05-20:
Thanks for the note. Please open a new issue for that on the DSF repo.