Duet3D/RepRapFirmware #402: Paneldue 7i does not connect with duet.

austaquio32, 2020-05-20:
good morning, I have a duet2 wifi with a paneldue 7i, with the version of reprap 3.01 rc11 and the paneldue works correctly, when updating to version 3.1.1 the paneldue does not connect with duet, I have it connected with the 4-wire cable, that could be happening.
ps: if i go back to rc11 it works correctly again.

T3P3, 2020-05-20:
Have you seen this in the Whats New file:

> All PanelDue users: the PanelDue connector (or IO_0 on Duet 3) is no longer dedicated to PanelDue, therefore if you connect a PanelDue to this port you must use the following command in config.g to enable it: M575 P1 S1 B57600. You can use baud rates other than 57600, however the IAP files all assume 57600 baud; therefore if you use another baud rate then PanelDue will not display firmware update progress.

https://github.com/Duet3D/RepRapFirmware/blob/v3-dev/WHATS_NEW_RRF3.md

giostark, 2020-05-20:
Hi, have you already added this line?
Release note 3.1.0:
https://github.com/Duet3D/RepRapFirmware/blob/v3-dev/WHATS_NEW_RRF3.md
> All PanelDue users: the PanelDue connector (or IO_0 on Duet 3) is no longer dedicated to PanelDue, therefore if you connect a PanelDue to this port you must use the following command in config.g to [...]

austaquio32, 2020-05-20:
That was the problem, that I did not stop reading the readme, sorry for the failure, thank you all.

closed 2020-05-20 by austaquio32