Duet3D/RepRapFirmware #403: BLTouch does not deploy after cancelled job.

icecube45, 2020-05-24:
on RRF3.1.1, duet 2 wifi, after cancelling a print, bltouch no longer deploys or retracts.
after board reset, bltouch deploys and retracts as expected.

Is this expected behavior or have I run into a 3.1.1 bug?

Carl-Higby, 2020-06-03:
I have the same issue may be a bug

icecube45, 2020-06-03:
I've been trying to report it on the forums but keep getting caught by the spam filter. 
Good to know it's not just me.

T3P3, 2020-06-03:
Specifically about the forum access issue please email me info@duet3d.com and will try and resolve that.

charames, 2020-07-02:
This may not be helpful, but I do not have that issue on RRF3.1.1 (Duet Maestro).
Maybe you could try resetting the BLTouch in stop.g (M280 P0 S160).  Note: yours may not be on P0 (adjust accordingly).

icecube45, 2020-07-02:
(oops forgot I made the report on github too)
Was able to solve this on forums at
https://forum.duet3d.com/topic/16791/3-1-1-bltouch-doesn-t-function-after-cancelling-a-job

closed 2020-07-02 by icecube45