Duet3D/RepRapFirmware #412: [RRF3.1.1] moving head using web inface causes stutter from time to time

boa-pe, 2020-06-13:
On the movie I am just clicking on +100 +100 -100 -100 in X axis.
From time to time there is additional "slow down" in the middle of second movement, like something is interfering with part responsible for motion controlling

https://www.youtube.com/watch?v=JCQg-M_0LxQ
First time at the beginning and second time at 0:57

Also reproducible with release 3.0 and 3.1.0

Nibbels, 2020-06-13:
I saw such, wenn the Z-Jerk was too low for the auto mesh leveling.
But your video does not show a full repeatability.

Is mesh leveling activated at this moment?

boa-pe, 2020-06-14:
No, mesh leveling is not enabled. I just homed XYZ, raised X a little bit, and played a little with X axis. That is all. Also I do not have this recorded but sometimes also uploading files during print causes a short "stutter" whet upload starts.

boa-pe, 2020-06-14:
Also I createst some gcode that does just same thing - X0 X100 X200 X100 X0 and so on, and is executed flawless. Seems there might be some interference between web iface handling and motion handling part.

boa-pe, 2020-08-13:
I saved a macro during a print, and print has been disturbed for the moment at the exact saving file event.
[image]

n8bot, 2020-08-13:
The small stutter when an upload starts might be an issue -- I think the code for that type of thing has been recently changed, so perhaps it is not exactly perfect yet.

However, the behaviour of repeated clicks on movement buttons through DWC (or PanelDue for that matter) causing a delay is expected. I believe the firmware will only execute four in a row, and won't accept any more commands until those are complete. Something like that.

closed 2023-05-04 by T3P3