Duet3D/RepRapFirmware #418: RRF 3.1.1 heater PID autotune always reporting as failed

zlowred, 2020-07-20:
Should be a minor regression, noticed on 3.1.1 but could be earlier.
Whenever I do PID autotune, like `M303 H1 S250` it seems to complete successfully, but issuing `M303` afterwards always says `Heater 1 tuning failed`. Same happens when I try to auto-tune bed heater so probably unrelated to my hotend/thermistor/heater cartridge.
```
7/20/2020, 2:27:53 PM | M303Heater 1 tuning failed
7/20/2020, 2:27:31 PM | Auto tune heater 1 completed in 278 sec Use M307 H1 to see the result, or M500 to save the result in config-override.g
7/20/2020, 2:24:20 PM | Auto tune phase 3, peak temperature was 269.2
7/20/2020, 2:24:08 PM | Auto tune phase 2, heater off
7/20/2020, 2:22:57 PM | Auto tune phase 1, heater on
7/20/2020, 2:22:51 PM | M303 H1 S250Auto tuning heater 1 using target temperature 250.0°C and PWM 1.00 - do not leave printer unattended
```
Diagnostics post tuning:
```
7/20/2020, 2:29:34 PM | M122=== Diagnostics === RepRapFirmware for Duet 2 WiFi/Ethernet version 3.1.1 running on Duet WiFi 1.02 or later Board ID: 08DDM-9FAM2-LW4S8-6JTDD-3SJ6P-9MXBY Used output buffers: 3 of 24 (12 max) === RTOS === Static ram: 27980 Dynamic ram: 95124 of which 564 recycled Exception stack ram used: 528 Never used ram: 6876 Tasks: NETWORK(ready,368) HEAT(blocked,752) MAIN(running,1912) IDLE(ready,80) Owned mutexes: WiFi(NETWORK) === Platform === Last reset 00:13:59 ago, cause: power up Last software reset at 2020-07-19 21:24, reason: User, spinning module GCodes, available RAM 7372 bytes (slot 3) Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x0440f000 BFAR 0xe000ed38 SP 0xffffffff Task MAIN Error status: 0 MCU temperature: min 32.4, current 33.0, max 37.9 Supply voltage: min 0.3, current 24.2, max 24.5, under voltage events: 3, over voltage events: 0, power good: yes Driver 0: standstill, SG min/max 0/148 Driver 1: standstill, SG min/max 0/132 Driver 2: standstill, SG min/max 0/170 Driver 3: standstill, SG min/max not available Driver 4: standstill, SG min/max not available Date/time: 2020-07-20 14:29:33 Cache data hit count 1447674344 Slowest loop: 346.64ms; fastest: 0.09ms I2C nak errors 0, send timeouts 0, receive timeouts 0, finishTimeouts 0, resets 0 === Storage === Free file entries: 10 SD card 0 detected, interface speed: 20.0MBytes/sec SD card longest read time 44.7ms, write time 0.0ms, max retries 0 === Move === Hiccups: 0(0), FreeDm: 169, MinFreeDm: 166, MaxWait: 56098ms Bed compensation in use: mesh, comp offset 0.000 === MainDDARing === Scheduled moves: 38, completed moves: 38, StepErrors: 0, LaErrors: 0, Underruns: 0, 0  CDDA state: -1 === AuxDDARing === Scheduled moves: 0, completed moves: 0, StepErrors: 0, LaErrors: 0, Underruns: 0, 0  CDDA state: -1 === Heat === Bed heaters = 0 -1 -1 -1, chamberHeaters = -1 -1 -1 -1 === GCodes === Segments left: 0 Movement lock held by null HTTP is idle in state(s) 0 Telnet is idle in state(s) 0 File is idle in state(s) 0 USB is idle in state(s) 0 Aux is idle in state(s) 0 Trigger is idle in state(s) 0 Queue is idle in state(s) 0 Daemon is idle in state(s) 0 Autopause is idle in state(s) 0 Code queue is empty. === Network === Slowest loop: 15.50ms; fastest: 0.00ms Responder states: HTTP(2) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0), 0 sessions HTTP sessions: 1 of 8 - WiFi - Network state is active WiFi module is connected to access point  Failed messages: pending 0, notready 0, noresp 0 WiFi firmware version 1.23 WiFi MAC address ec:fa:bc:02:1e:41 WiFi Vcc 3.35, reset reason Unknown WiFi flash size 4194304, free heap 22608 WiFi IP address 192.168.86.100 WiFi signal strength -65dBm, reconnections 0, sleep mode modem Socket states: 0 0 0 0 0 0 0 0
```

This does not prevent proper PID tuning, just `M303` reporting seems to be broken.

zlowred, 2020-07-20:
Before tuning (post reset) `M303` correctly says `No heater has been tuned yet`, and it also correctly reports when autotuning in progress (don't remember exact wording, but something saying PID tuning in progress)

zlowred, 2020-07-21:
config.g in case it matters
```
; Configuration file for Duet WiFi (firmware version 3)
; executed by the firmware on start-up
;
; generated by RepRapFirmware Configuration Tool v3.1.3 on Sat Jul 04 2020 19:21:49 GMT+0800 (Singapore Standard Time)

; General preferences
G90                                                ; send absolute coordinates...
M83                                                ; ...but relative extruder moves
M550 P"Scraps"                                     ; set printer name
M669 K1                                            ; select CoreXY mode

; Network
M552 S1                                            ; enable network
M586 P0 S1                                         ; enable HTTP
M586 P1 S0                                         ; disable FTP
M586 P2 S0                                         ; disable Telnet

; Drives
M569 P0 S0                                         ; physical drive 0 goes forwards
M569 P1 S0                                         ; physical drive 1 goes forwards
M569 P2 S1                                         ; physical drive 2 goes forwards
M569 P3 S1                                         ; physical drive 3 goes forwards
M569 P4 S1                                         ; physical drive 4 goes forwards
M584 X1 Y0 Z2 E3:4                                 ; set drive mapping
M350 X16 Y16 Z16 E16:16 I0                         ; configure microstepping with interpolation
M92 X160.00 Y160.00 Z1600.00 E2825.00:2825.00      ; set steps per mm
M566 X3000.00 Y3000.00 Z300.00 E300.00:300.00      ; set maximum instantaneous speed changes (mm/min)
M203 X18000.00 Y18000.00 Z1200.00 E3000.00:3000.00 ; set maximum speeds (mm/min)
M201 X5000.00 Y5000.00 Z700.00 E2500.00:2500.00    ; set accelerations (mm/s^2)
M906 X1800 Y1800 Z2000 E300:300 I30                ; set motor currents (mA) and motor idle factor in per cent
M84 S30                                            ; Set idle timeout

; Axis Limits
M208 X0 Y0 Z0 S1                        	       ; set axis minima
M208 X317 Y380 Z351.94 S0                             ; set axis maxima

; Endstops
M574 X1 S1 P"!xstop"                                ; configure active-high endstop for low end on X via pin xstop
M574 Y1 S1 P"!ystop"                                ; configure active-high endstop for low end on Y via pin ystop
M574 Z2 S1 P"!zstop"                                ; configure active-high endstop for high end on Z via pin zstop

; Z-Probe
M950 S0 C"exp.heater7"                             ; create servo pin 0 for BLTouch
M558 P9 C"^zprobe.in" H10 F300 T18000              ; set Z probe type to bltouch and the dive height + speeds
G31 P500 X0 Y0 Z2.5                                ; set Z probe trigger value, offset and trigger height
M557 X3:303 Y8:308 S30                             ; define mesh grid

; Temp sensors
M308 S0 P"bedtemp" Y"thermistor" T100000 B3950     ; configure sensor 0 as thermistor on pin bedtemp
M308 S1 P"spi.cs1" Y"rtd-max31865" R430 F50 W2 A"Pt1000 1"    ; configure sensor 1
M308 S2 P"spi.cs2" Y"rtd-max31865" R430 F50 W2 A"Pt1000 2"     ; configure sensor 2
M308 S3 P"spi.cs3" Y"rtd-max31865" R430 F50 W2 A"Pt100 1"    ; configure sensor 3
M308 S4 P"spi.cs4" Y"rtd-max31865" R430 F50 W2 A"Pt100 2"    ; configure sensor 4
M308 S5 Y"mcu-temp" A"MCU"						   ; configure MCU temp sensor
M308 S6 Y"drivers" A"Drivers" 				       ; configure stepper drivers temp sensor

; Heaters
M950 H0 C"bedheat" T0 Q2                           ; create bed heater output on bedheat and map it to sensor 0
M307 H0 B0 S1.00                                   ; disable bang-bang mode for the bed heater and set PWM limit
M140 H0                                            ; map heated bed to heater 0
M143 H0 S180                                       ; set temperature limit for heater 0 to 180C
M950 H1 C"e0heat" T1                               ; create nozzle heater output on e0heat and map it to sensor 1
M307 H1 B0 S1.00                                   ; disable bang-bang mode for heater  and set PWM limit
M143 H1 S450                                       ; set temperature limit for heater 1 to 450C
M950 H2 C"e1heat" T2                               ; create nozzle heater output on e1heat and map it to sensor 2
M307 H2 B0 S1.00                                   ; disable bang-bang mode for heater  and set PWM limit
M143 H2 S450                                       ; set temperature limit for heater 2 to 450C

[... 11 lines cut ...]
G10 P0 R0 S0                                       ; set initial tool 0 active and standby temperatures to 0C
M563 P1 S"Secondary" D1 H2 F0                      ; define tool 1
G10 P1 X100 Y100 Z0                                    ; set tool 1 axis offsets
G10 P1 R0 S0                                       ; set initial tool 1 active and standby temperatures to 0C
M563 P2 S"BLTouch" F-1                             ; define tool 2
G10 P2 X0 Y0 Z0                                    ; set tool 2 axis offsets
G10 P2 R0 S0                                       ; set initial tool 2 active and standby temperatures to 0C
; M563 P3 S"Probe" F-1                               ; define tool 3
; G10 P3 X0 Y0 Z0                                    ; set tool 3 axis offsets
; G10 P3 R0 S0                                       ; set initial tool 3 active and standby temperatures to 0C

M950 S1 C"exp.heater6"							   ; set up servo pin for the tool changer
M280 P1 S90

; Custom settings are not defined

; Miscellaneous
; M575 P1 S1 B57600                                ; enable support for PanelDue
M501                                               ; load saved parameters from non-volatile memory
M911 S22 R23 P"M913 X0 Y0 G91 M83 G1 Z3 E-5 F1000" ; set voltage thresholds and actions to run on power loss

M207 P0 S0.4 F2000 Z0.1							   ; FW retraction

M81												   ; Power off -> Show ATX controls in DWC
```

T3P3, 2023-05-03:
should be resolved, please test with 3.4.5 and re-open if not.

closed 2023-05-03 by T3P3