Duet3D/RepRapFirmware #421: Part 20% over weight predicted by slicer

jschall, 2020-08-28:
So, PrusaSlicer said my part would weigh something like 7.4 grams, and it came out weighing 8.7 grams, and there was definitely evidence of overextrusion.

I've checked:
- esteps calibration (slight underextrusion)
- filament density (correct in slicer)
- filament diameter (correct in slicer)

RRF's reported extruder position at the end of the print matched the slicer's predicted filament length.

Gcode file: [prop_pc.gcode.txt](https://github.com/Duet3D/RepRapFirmware/files/5139928/prop_pc.gcode.txt)

However, I also printed a 15mm cube with 70% rectilinear infill, and it came out underweight! One of the main differences I can think of is that the cube required no retractions. So, next I'll try printing 2 of them and turning on retraction on layer change...

Could it be some kind of interaction between retraction and pressure advance?

jschall, 2020-08-28:
[image]

Printed result
PolyMax PC
E3D Tool Changer
RRF 2.03 (2019-06-13b2)

jschall, 2020-08-28:
Also tried using slicer retraction and using firmware retraction with the same result. Haven't tried no retraction yet.

jschall, 2020-08-28:
I do see that compensation is conditional on direction in the code. It is a bit much to grok right now, but I'll just try printing without any retraction, and I'll also try printing without any pressure advance if I can get the print to succeed like that.

jschall, 2020-08-28:
No retract print came in overweight, trying no pressure advance.

At some point I'll measure out the length of filament used just to be sure I'm not crazy...

jschall, 2020-08-28:
Changed 2 things:
- Switched to absolute extrusion because I noticed PrusaSlicer was set to relative
- Turned off pressure advance

The result is looking good so far. After it is done, assuming it weighs the right amount, I'll turn pressure advance on and see which was the culprit.

jschall, 2020-08-28:
Yep, so, with the same file and pressure advance turned on, severe overextrusion.

This is despite the web interface showing the extruder position as almost exactly the correct extrusion amount.

jschall, 2020-08-28:
Now running again with extremely low extruder jerk, acceleration and max speed, to ensure that there are no missed steps.

jschall, 2020-08-28:
Still overextrusion. So, pressure advance somehow causes 20% overextrusion on this file.

Billtopia, 2020-08-29:
The weight in your slicer is an estimate based on the mass per cubic centimeter of the filament that **you** have to set in the slicer. It is just done with the volume of extruded plastic multiplied by the **user supplied** mass of the plastic. 

Pressure advance does not cause overextrusion either... It doesn't change how much filament is extruded but rather the extrusion rate over the entire extrusion move. If not calibrated correctly it can make prints look worse than without but you are still extruding the same amount of material either way. 

If your e-steps are incorrect you should properly calibrate them first.

If your e-steps are correct you have to check the extrusion multiplier in your slicer to correct the overextrusion. Every filament is slightly different, from brand, color, material, quality... so you need to do an extrusion multiplier calibration print for your current filament and set the multiplier to the proper value.

jschall, 2020-08-29:
> Pressure advance does not cause overextrusion either... It doesn't change how much filament is extruded but rather the extrusion rate over the entire extrusion move.

Are you real sure about that one?

Steps as counted by logic analyzer connected to step and direction pins:

Without pressure advance:
fwd_count 1942964
rev_count 961423
net_count 981541
mm 2484.913924

Same file with pressure advance set to 1.0:
fwd_count 6334445
rev_count 5157566
net_count 1176879
mm 2979.440506

With regard to the rest of your post, if you actually read what I'm saying you'd see I've controlled for filament diameter, filament density, etc etc etc. Why even reply without reading and comprehending first?

StephenRC, 2020-09-01:
What was the microstepping?  The Duet 2 and later can have variable
microstepping.

On Sat, 29 Aug 2020 08:09:36 -0700, jschall <notifications@github.com>
had a flock of green cheek conures squawk out:

> Are you real sure about that one?
>
> Steps as counted by logic analyzer connected to step and direction pins:
>
> Without pressure advance:
> fwd_count 1942964
> rev_count 961423
> net_count 981541
> mm [...]

T3P3, 2021-02-22:
@jschall do you still see an issue with RRF 3.2?

closed 2023-05-04 by T3P3