Duet3D/RepRapFirmware #423: zprobe

gadget808, 2020-09-06:
After installing 3.1.1 my z probe is now reversed.  It show 1000 when not triggered and 0 when triggered.

Here is the Z Probe config.  Z Probe is an EZ ABL Pro:

M574 Z2 S2                                                       ; set sendstops controlled by probe ezabl pro
M558 P5 C"io3.in" R0.4 H5 F1200 T6000 I1      ; set Z probe type to effector and the dive height + speeds
M950 S0 C"io3.out"
G31 P1000 X-45 Y20 Z1.87                                ; set Z probe trigger value, offset and trigger height
M557 X50:300 Y35:340 S50                               ; define mesh grid

gadget808, 2020-09-07:
I have tried moving the I1 around and nothing I do will change the reverse on the probe.

dc42, 2020-09-07:
Please read the documentation for M558 in RRF3 and if necessary use the forum support.

closed 2020-09-07 by dc42

gadget808, 2020-09-07:
Sorry forgot to come back and post that I figured out the needed addition

M574 Z1 S2                                             ; set sendstops controlled by probe ezabl pro
M558 P5 C"!io3.in" R0.4 H5 F1200 T6000               ; set Z probe type to effector and the dive height + speeds
M950 S0 C"io3.out"
G31 P500 X-45 Y20 Z1.87                               ; set Z probe trigger value, offset and trigger height
M557 X50:300 Y35:340 S50