Duet3D/RepRapFirmware #435: Pressure Advance set by job: Simulation time incorrect [RRF 3.1.1]

seeul8er, 2020-10-10:
Setting the PA value in the actual print file causes the simulation time to be too low. If PA is set within the RRF config the simulated times are way more accurate. I tend to have a PA value per filament so setting the value in the job file makes sense.

[Example.zip](https://github.com/Duet3D/RepRapFirmware/files/5358906/Example.zip): PA 0.65
- PA in .gcode file -> Sim time: 53min
- PA in .config file -> Sim time 1h24min

Hardware/Software: RepRapFirmware for Duet 2 WiFi/Ethernet version **3.1.1** running on Duet WiFi 1.02 or later

michele-bertoni, 2020-10-21:
What about defining PA in the config.g of your filament? You will have different PA for each filament and the simulation time should be accurate

T3P3, 2023-05-04:
Please open a forum discussion if this is still an issue in RRF 3.4.5

closed 2023-05-04 by T3P3