Duet3D/RepRapFirmware #437: Abort command in load.g or unload.g filament macros changes filament, as it was correctly loaded/unloaded

michele-bertoni, 2020-10-16:
If an abort command is called when loading or unloading a filament (both with M701 or M702 and with the Duet Web Control), Duet will then change the filament name of the associated tool, as the filament was correctly loaded or unloaded. A very strange behavior happens, instead, when an error (such as G0/G1: insufficient axes homed) is thrown: in this case the filament name is not changed. The problem happens both if the abort command is inside load.g/unload.g file and if it is inside a nested macro called by load.g/unload.g.

zombiRon, 2020-10-16:
forget abort, you can't even change filament without a reset

michele-bertoni, 2020-10-17:
> forget abort, you can't even change filament without it resetting

You mean the error present in latest beta release? It has already been fixed for the next beta release

zombiRon, 2020-10-17:
Yes, so how are you testing this without it resetting?

michele-bertoni, 2020-10-17:
> Yes, so how are you testing this without it resetting?

I'm using latest working release and, since I haven't found anyone reporting this bug, I assumed that it wasn't fixed yet

zombiRon, 2020-10-17:
you're running 3.1.1?

michele-bertoni, 2020-10-17:
Yes, 3.1.1, last stable release

michele-bertoni, 2020-10-23:
I found out that if the tool is deselected (with T-1) at the time the load/unload macro ends, the filament won't change its state. So, if you do T-1 before calling abort, it works fine.

skezo, 2023-04-15:
This issue is present in `release 3.4.5`.
If you run `abort "Cancel load"` inside `/filaments/<filament>/load.g` or inside a submacro of `load.g` the Filament will change state.

T3P3, 2023-07-19:
Closed as M99 is the correct approach - see the forum thread.

abort commands are not supported at this point in system macros.

closed 2023-07-19 by T3P3