Duet3D/RepRapFirmware #438: Tool changes in mode FFF do not honor Z axis limits

wilriker, 2020-10-16:
If a tool change is performed in mode FFF (and possibly Laser) it can exceed Z axis limits if the tool's Z offset is negative at and the head is at the top end or tool offset is positive with head at the low end respectively.

zombiRon, 2020-10-23:
Just a thought...
I'm not sure the best case here is to cancel the job. I use a z-offset when using the laser toolhead which I have calibrated to the ideal focal distance. This gives me a lower max Z than when in FFF mode.

I have my z endstop at the farthest end of the axis so when homing I don't have to worry about anything colliding at the end of a job, I use z probe to set z=0 before an FFF job. Now if Z is home (max position) and I select T1 which is my laser, the bed attempts to go beyond the physical limit and causes motor to screem for a bit. At this point all I have done is select a tool, my tool home positions can't collide with the bed they're physically far enough away.

Can we not update the current Z position (equiv of a G92 Z) rather than physically move the Z-axis, then the next move command would be able to calculate the new position correctly and move the Z Axis accordingly without the risk of crashing the head or exceeding the physical limit of the axis??

wilriker, 2020-10-23:
Depending on the machine and how powerful the motors are it could already be a big problem if the axis tries to exceed the top limit. It will certainly not crash with the bed but it might bend or break other parts. So this definitely needs to check both ends. And while simply adjusting the Z position seems like a good middleground solution try to imagine the following scenario:
Machine is a CNC tool changer with a revolving toolhead to change tools. Currently the machine is at Z max with stock material just fitting under the current cutter. Next up is to load a longer cutter that will just not fit between the current location and stock. Suppose this machine does not have a means to reliably detect failed (physical) tool changes and the head is now in a non-perpendicular position with the cutter at an angle pressed against stock. Now start spindle at 30.000 RPM - you better wear some body-armor...

Short: there is only one really safe solution and that is aborting the job. I understand that this can be annoying in some situations where it would have been perfectly safe to just do it but there is always this one situation where it is too dangerous.

zombiRon, 2020-10-23:
What moron is overloading a heavy duty machine with stock that "only just fits"??

wilriker, 2020-10-23:
It was just an example of what can go wrong. You can possibly construct other scenarios that will end equally bad for the tool, the machine or worst case the operator. All I want to say is: abort is the only safe option because the firmware cannot know which "200 pound shop gorilla" is currently in charge.

zombiRon, 2020-10-23:
aww dude, you're just shortening next years darwin awards contenders list :)

zombiRon, 2020-10-23:
In case anyone else has the same issue I have given this change, my solution is to define G55 for the laser mode, which means this change will respect and police the physical machine limits, but allows me to move the coordinate system without abusing the tool offset.

commit 55b6380c (2020-10-29): Abort tool change and possibly running job if applying tool Z offset
would exceed Z min/max (fixes #438)

Extend aborting a job by retracting the probe and running abort.g
commit 049b381f (2020-11-06): Abort tool change and possibly running job if applying tool Z offset would exceed Z min/max (fixes #438)
commit 927f1ec9 (2020-11-06): Abort tool change and possibly running job if applying tool Z offset would exceed Z min/max (fixes #438) (#442)
closed 2020-11-06 by wilriker