Duet3D/RepRapFirmware #443: [Feature Request] Add "input shaping" to RRF

T3P3, 2020-10-22:
This is now a feature in Klipper (https://www.klipper3d.org/Resonance_Compensation.html) that appears to get better results than [DAA](https://duet3d.dozuki.com/Wiki/Gcode#Section_M593_Configure_Dynamic_Acceleration_Adjustment)

Apparently based on this paper:
https://www.researchgate.net/profile/Peter_Meckl/publication/224606974_Input_shaping_for_nonlinear_systems/links/548781470cf268d28f0724e7/Input-shaping-for-nonlinear-systems.pdf

If not appropriate for RRF then I will move this request to DSF.

jlipavsky, 2020-12-26:
very interested to see some progress ade on this.  Have a corexy with a duet3 and a pi4/4gb and the accel board needed, would love to have the machine tune some settings by themselves and i dont see it stopping there either, whats to say that it couldnt monitor for incorrect movement

astorun, 2020-12-28:
Definitely want to see it as a feature on Duet 3 series as well! I have a large format corexy (720x720x400) printer that prints great but will benefit from it especially on ghosting.

astorun, 2021-02-11:
Any updates on this one?

dc42, 2021-02-12:
Planned for RRF 3.4.

mylife4aiurr, 2021-06-11:
This pull request was featured in a teaching tech youtube video:
https://youtu.be/er7q-CJL1lc

Was able to get the only compatible sensor (LIS3DH) wired and talking to rrf 3.3beta. Im looking forward to trying out your implementation. 
Picking the right orientation configuration was hard. I'm not certain I have the right orientation configured. Seeing the auto-configuration in Klipper is mind-blowing. If the user can wire and mount the sensor it would awesome if firmware took it from there.

joaobarros, 2021-06-12:
Funny you mention that video, I'm a beta tester for RatRrig and it shows a video (0:57) of my V-Core3 (CoreXY) doing a kinematic bed "dance" (custom config.g and macro) with a Duet3 6HC
Amongst the beta testers it's a small amount of Duet users vs a horde of Klipper users.

Nonetheless I was curious about Klipper, specifically the feature of input shaping.
I'm currently beta testing a new small bed slinger, the V-Minion with an old Duet2 Wifi running Klipper.
I intend to test it also with RRF 3.4 to be able to compare both implementations and report back :)

dc42, 2022-03-29:
Implemented in RRF 3.4.0.

closed 2022-03-29 by dc42