Duet3D/RepRapFirmware #444: Ideamaker, Reprap does not detect slicer print time and filament usage

alessg1, 2020-10-24:
Reprap firmware does not find ideamaker(4.0) print time and filament usage.
My current setup is:
Board: Duet 2 WiFi (2WiFi)
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 3.2-beta2 (2020-10-05b2)
Duet WiFi Server Version: 1.24beta2-05b1 

Ideamaker puts print time and filament usage at the end of the gcode.

These are the last lines of gcode
;
;Print Time: 4897
;Material#1 Used: 4493.7
;Material#2 Used: 0.0
;Material#1 Cost: 0.40
;Material#2 Cost: 0.00

pfn, 2021-01-12:
https://github.com/Duet3D/RepRapFirmware/blob/3.2/src/Storage/FileInfoParser.cpp#L689

Looks like it's searching for " Used:\t" while the string is " Used: "

pfn, 2021-01-12:
Looking at this further, seems to be sizeToScan that's preventing reaching the end of the file to parse.

T3P3, 2023-05-04:
please reopen this issue if it still persists in RRF 3,4,5 with the latest version of IdeaMaker.

closed 2023-05-04 by T3P3