Duet3D/RepRapFirmware #445: Feature Request: Custom M557 Mesh Probing Grid

garethky, 2020-10-25:
Magnetic beds with removable spring steel build surfaces have become quite popular. One difficulty with using this kind of build surface is mesh bed probing with an inductive sensor. Its critical that the sensor probe in spots not influenced by the magnets in the bed to get an accurate reading. The regular grids supported by `M557` currently don't always allow magnets to be avoided with the desired probing density or extents.

This is a request to allow `M557` to be supplied with x/y coordinate pairs to specify the exact probing grid locations. E.g. they could be passed in from a file with contents such as this:

```
x15 y6
x20 y20
x20 y32
x20 y40
x15 y56
...
```

This would allow a machine builder or bed manufacturer to specify a grid of safe probing locations based on their knowledge of the magnet locations in the bed.

(I've skimmed the code and I think it would be something of an overhaul of the mesh probing code to support this. The height map file doesn't store the x/y locations probed which would require an update with backwards compatibility concerns. Likely, a regular grid would still be required to do on the fly height adjustment and that grid would have to be interpolated from the real probe points.)

P-C-R, 2020-10-27:
You could achieve this by probing with g30 and the P parameter

garethky, 2020-10-27:
> You could achieve this by probing with g30 and the P parameter

That's what I originally thought too.

`G30` [Single Z Probe](https://duet3d.dozuki.com/Wiki/Gcode#Section_G30_Single_Z_Probe) it not integrated with `G29` [Mesh Bed Probe](https://duet3d.dozuki.com/Wiki/Gcode#Section_G29_Mesh_bed_probe). Mesh bed probing allows for the surface of the be bed to be uneven, cupped, warped or twisted and applies a corrective mesh to Z as the print head moves. G30 limits you to a flat correction plane described by the average of, at most, 5 points in a Cartesian printer.

From the G30 documentation:
_From version 1.09, the number of factors may be 3, 4 or 5 when doing old-style auto bed compensation on a Cartesian or CoreXY printer, and 3, 4, 6, 7, 8 or 9 when doing auto calibration of a Delta printer._

I believe this is the relevant source code reference for G30 correction: [RandomProbePointSet::SetProbedBedEquation](https://github.com/Duet3D/RepRapFirmware/blob/90c6b876d29e8de72da5b044327d2bd8ac073279/src/Movement/BedProbing/RandomProbePointSet.cpp#L95). This code actually drops a [warning message](https://github.com/Duet3D/RepRapFirmware/blob/90c6b876d29e8de72da5b044327d2bd8ac073279/src/Movement/BedProbing/RandomProbePointSet.cpp#L170) to stop using G30 and switch to G29 as G30 may be deprecated in a future release.

astorun, 2020-12-24:
I upvote for that too! Ability to declare custom points will definitely works for me especially. I am using 9 seperate buildplates instead of one large one on my oversized print surface.

garethky, 2021-01-24:
I have worked out a way to accomplish this with the SBC, Python and [ExecOnMCode](https://github.com/wilriker/execonmcode). I'm still testing but so far its showing promise. Once I'm confident that I don't have anything major wrong I'll release the Python code and a guide to setting it all up.

I still need to work out how I'm going to handle per-build-plate Z-offsets. I have a way to store the baby stepping value per-tool in my toolchanger code. But I need to generally modify the tool's z-offset for each build plate. I think this will have to be a separate extension to DWC.

garethky, 2021-02-01:
Here is the result: [](https://github.com/garethky/rrf-mesh-level)

I guess the question for RRF is: could be this Radial Basis Functions technique be used in the firmware to compute the bed shape. Even the stock mesh bed leveling would benefit from being up-sampling from an RBF interpolation. It wouldn't have to impact runtime performance because we can continue to use the liner interpolation code for that. We would just need to port the RBF code from Python to C and add support for arbitrary point sets for G29 to fully solve this.

dc42, 2021-02-01:
@garethky, can you provide a photo or diagram of a magnetic bed of the type that you have in mind, that shows where the magnets are, so that I can better understand why a rectangular grid is not suitable?

garethky, 2021-02-02:
@dc42 sure thing, this is my bed:
[image]

This is the probing pattern that I'm using with it with rrf-mesh-level. Each small cylinder is a probe location, the large circles are the magnet exclusion zones:

[image]

There are at least 2 companies making magnetic beds for printers that are typically equipped with the Duet boards:
Mandala Rose Works: [RailCore](https://www.mandalaroseworks.com/shop/railcore/railcore-ultraflat-magbed), [Jubilee](https://www.mandalaroseworks.com/shop/mrw-products/jubilee-magbed) and [E3D Tool Changer](https://www.mandalaroseworks.com/shop/mrw-products/magbed-for-e3d-toolchanger)
713 Maker: [Railcore](https://713maker.com/railcore/railcore-ii-250zl/railcore-ii-250zl-magnetic-heated-bed), [Jubilee](https://713maker.com/jubilee/jubilee-magnetic-heated-bed)

This is the Prusa MK3S bed which is probably the most widely distributed magnetic bed:
[image]

The Prusa community found that a 7x7 probing grid gave the best first layer performance on the 210mm x 210mm sized bed. That's 1 probing point per ~26mm. 11 of those locations are impacted by magnets. Prusa ended up solving this by ignoring those 11 locations, see [Magnet comp.](https://help.prusa3d.com/en/article/mesh-bed-leveling_112163#mesh-bed-leveling-settings). This isn't ideal as it reduces the sample density in those areas and wastes time sampling points that wont be used. This would be the simplest change that would work though.

The magnetic PCB heater bed design has a major cost advantage over the aluminum beds but we don't see them on the market. One of the reasons is the inability to effectively probe them. The aluminum beds have better flatness tolerances so they can get by with fewer probed points. Irregular grid probing would help enable cheap heated magnetic PCB beds for these printers.

The only other tool that I am aware of that tries to solve this problem for the Duet/RRF is a [google docs spreadsheet](https://docs.google.com/spreadsheets/d/19sg3DF3PqzK8-KJ13Eh1PQMJRK1Q9LHXkGnUAmX1ld8/) but it will only work for 2x2 probing grids which is not true mesh leveling.

dc42, 2021-02-16:
Implementing a totally custom grid would need additional memory and complicate the algorithm significantly. How about using a regular grid but having the option to skip or ignore some points?

P-C-R, 2021-02-16:
@dc42 that would.be the best Option! With that you can easily make a custom
Grid. Or make online Tool for.it! Most people do Not need a custom one

garethky, 2021-02-16:
Skipping points works.

My solution was to up-sample the custom grid into the regular one so there is no additional runtime memory or overhead. But I'm not about to dig through that Python library and backport it to C.

T3P3, 2023-05-04:
This is now supported in RRF 3.5.0 using G29 S4
https://docs.duet3d.com/en/User_manual/Reference/Gcodes#parameters-8