Duet3D/RepRapFirmware #452: Transition from Stable 3.1.1 to 3.2.0-beta4 fails on upgrade (RPi)

tdcox, 2020-11-28:
I attempted to install the beta version of the firmware using the following commands:

```
pi@toolchanger:~ $ wget -q https://pkg.duet3d.com/duet3d-unstable.list
pi@toolchanger:~ $ sudo mv duet3d-unstable.list /etc/apt/sources.list.d/duet3d-unstable.list
pi@toolchanger:~ $ sudo chown root:root /etc/apt/sources.list.d/duet3d-unstable.list
pi@toolchanger:~ $ sudo rm /etc/apt/sources.list.d/duet3d.list
pi@toolchanger:~ $ sudo apt-get update
pi@toolchanger:~ $ sudo apt-get upgrade
```

Pt install proceeded normally until I hit the following, then it failed:
 
```
Setting up reprapfirmware (3.2.0-beta4-1) ...
Sending update request to DCS... Error: Failed to send update request
Unhandled exception. System.Threading.Tasks.TaskCanceledException: A task was canceled.
   at DuetAPIClient.BaseConnection.PerformCommand[T](BaseCommand command, CancellationToken cancellationToken) in /home/christian/Duet3D/DuetSoftwareFramework/src/DuetAPIClient/BaseConnection.cs:line 188
   at DuetControlServer.Program.CheckForAnotherInstance() in /home/christian/Duet3D/DuetSoftwareFramework/src/DuetControlServer/Program.cs:line 306
   at DuetControlServer.Program.Main(String[] args) in /home/christian/Duet3D/DuetSoftwareFramework/src/DuetControlServer/Program.cs:line 84
   at DuetControlServer.Program.<Main>(String[] args)
Aborted
```

This is a new system that has a single Duet 3 board linked to a Raspberry Pi 4 and was previously working with minor issues under 3.1.1.

tdcox, 2020-11-28:
I rebooted and executed:

`sudo apt-get --reinstall install duetcontrolserver duetruntime duetsd duetsoftwareframework duettools duetwebcontrol duetwebserver libjavascriptcoregtk-4.0-18 libwebkit2gtk-4.0-37 reprapfirmware rpi-eeprom`

All appears to have installed correctly the second time around.

dc42, 2020-11-30:
I'm glad you solved it.

closed 2020-11-30 by dc42

tdcox, 2020-11-30:
The original bug still exists, I have just posted a workaround.

dc42, 2020-11-30:
This is probably a DCS issue, not a RRF issue.