Duet3D/RepRapFirmware #453: In marlin compatibility, missing' \n' after "OK" response when M20 command is used.

Subcircuit, 2020-12-02:
In marlin compatibility, **missing '\n'** after "OK" response when M20 command is used.   
FIRMWARE_VERSION: 3.1.1 
```
    Begin file list 
    text.gcode 
    End file list 
    ok (here missing '\n') 
```

Subcircuit, 2020-12-02:
Sending M20 from panelDUE found the problem, I tested the usb port and it is working.

dc42, 2020-12-02:
PanelDue does not require a terminating \n so it is omitted from most responses. If you put the aux port into raw mode using M575 (only available in recent RRF3 versions) then the \n should be present.

Subcircuit, 2020-12-03:
@dc42 Sorry, my description not clear.  
I'm using M575 P1 S2 B115200 with marlin compatibility mode and send m20  without ’\n‘ response at the end , but other gcodes (m104, m106, m119, m105, m21 etc.) do have ‘\n’!  
This prevents me from parsing the M20 response correctly.

dc42, 2020-12-03:
Please try RepRapFirmware 3.2beta4.1. The code to generate responses was changed in that version, so it may be fixed already.

Subcircuit, 2020-12-04:
Hi, I tested 3.2beta4, 3.2beta4.1 and still have the problem.

Subcircuit, 2020-12-21:
@dc42  
Before that I was testing on Duet 2 wifi board.
Today I tested Duet3 6HC firmware version 3.1.1 , there is no such problem. (All test environments are the same except for the board).

dc42, 2020-12-21:
Can you try firmware 3.2RC1 on a Duet WiFi?

Subcircuit, 2020-12-22:
Oh great, it's working. Thank u

closed 2020-12-23 by Subcircuit