Duet3D/RepRapFirmware #454: Duet Wifi 2 wifi server 1.25 issue

cdedwards, 2020-12-06:
I've been attempting to upgrade to the newer version of duet Wifi server ver 1.25. It will install and resets the  duet, but at that point it doesn't connect and gives an error of SPI transfer in progress if you attempt to M552 S-1, M552 S0 M587 S"*" in an attempt to restore networking. Forcing 1.23 using M997 S0:1  installs the older version and network works great from there.
One thing I noticed is that there seems to be a rather large size differance between the 2 files. 18kb might not seem like alot, but it's only a 272k file on 1.23 and 255kb on 1.25

dc42, 2020-12-07:
1. Did you try installing firmware 1.25beta0 more than once? The firmware upload protocol used by the ESP8266 unfortunately does not include any error checking, and I have seen cases where flashing the firmware a second time worked.
2. On a new WiFi module it is common for the first attempt to use M587 to add the first SSID to fail with SPI timeout. The second attempt succeeds.
3. M587 S"*" is not a valid command.

stavrosg, 2022-03-17:
Tripped over this thread during the same problem, solution worked perfectly. Thank you!

closed 2022-03-29 by dc42