Duet3D/RepRapFirmware #456: Setting Pressure Advance in Filament Config

tdcox, 2020-12-08:
I have configured retraction and pressure advance in config.g for each of my filament definitions as follows:

```
M207 P{state.currentTool} S8.0 F2400 T2400 Z0.075		; Firmware retraction
M572 D{tools[state.currentTool].extruders[0]} S0.4		; Pressure advance

In tpost*.g, I have `M703`.

I'm not getting any errors reported, but if I run `M207` or `M572` whilst printing, I see the default values from my startup config.

tdcox, 2020-12-08:
Version: 3.2.0-beta4

garethky, 2021-01-24:
The params to M572 were not correctly responding to expressions. See: https://forum.duet3d.com/topic/20788/m572-s-d-parameter-doesn-t-process-expressions

T3P3, 2023-05-04:
fixed in RRF 3.4

closed 2023-05-04 by T3P3