Duet3D/RepRapFirmware #466: Define more the 4 probes via M558

eehusky, 2021-01-17:
Not sure this qualifies as a bug but I thought this might be the best place to ask this.

What I would like to do is increase the number of probes that can be defined via M558 but I havent been able to track down in the source code where the max number is defined at.  I realize this might be a pretty niche use and dont expect you guys to do this for me, just point me in the right direction Ill take it from there.  Currently running this firmware version in standalone mode if it matters.  

Duet Firmware Vesion: RepRapFirmware for Duet 3 MB6HC 3.1.1 (2020-05-19b2)

I would prefer not to change from this baseline if i can help it (unless updating wont muck with my web controller updates)

---

Some more background for those who might be curious why I want to do this...I am using my Duet 3 board as a CNC controller, the firmware is completely stock at the moment but I have a pretty heavily modded web controller.  I am currently using the following gcode macros to handle tool switching and to properly activate the "spindle" model with the active tool.  Each tool # has a probe defined along with it to store the diameters of the tools when doing X/Y probing.  I've been using this for awhile and everything is working great except I just ran into a problem when I tried to add tool number 4 and received an error message telling me the K parameter is too high.

```
;config.g
M563 P0 D0 H0 F0 S"0.125 Probe"                      ; define tool 0
M558 K0 P8 F200 I0 R0.25 T500 A3 S0.02 C"nil"  ; Define probe, but dont allocate pin

;tfree.g
M453 S0 C"io8.out" R0 F0 T-1 									; Clear Tool Selection for spindle
M558 K{state.currentTool} P8 F200 I0 R0.25 T500 A3 S0.02 C"nil"		; Free Probe Pin

;tpost.g
M453 S0 C"io8.out" R0 T{state.currentTool}                              ; Assign tool to spindle (for duet web)
M558 K0 P8 F200 I0 R0.25 T500 A3 S0.02 C"!io6.in"	        ; Redefine/Acquire Probe Pin
G31 K0 X-1.5875 Y-1.5875 Z0 P1						; Assign X,Y offsets of probe (these get cleared when probe is deselected so it needs to be done again)

; When probing I can do this to correct for the radius of the end mill
; ...Other probing setup...
; Set current position to thickness of puck as well as X,Y radius of probe tool and 2mm above puck.
G1 X{10 - sensors.probes[state.currentTool].offsets[0]} Y{10 - sensors.probes[state.currentTool].offsets[1]} Z2
```

This is essentially implementing an extra tool parameter defining diameter...and I cant use tool offsets for this because the center of the tool is the center of the spindle and storing the offsets in the tool definitions will botch that up (I tried).  Its not terribly elegant but I wanted to implement this without cracking open an editor and i almost made it haha.

Thanks for all the work and help!
-Brandon

dc42, 2022-03-29:
How many Z probes do you need?

T3P3, 2023-05-04:
Cutter radius compensation as a feature request is here:
https://github.com/Duet3D/RepRapFirmware/issues/755

Please open a discussion on the forum if you still need to increase the maximum number of Z probes beyond 4.

closed 2023-05-04 by T3P3