Duet3D/RepRapFirmware #469: Support Multi-Axis Screw Mapping

MZachmann, 2021-01-29:
This is an enhancement request for multi-axis screw mapping. That is the ability to have piece-wise linear adjustment to position (of both a moving axis and any dependent axes) based on a lookup table. 

The most common example is to correct for inaccurate leadscrews (one dependent axis). In my case it adjusts for not-perfectly-straight and not-perfectly-coplanar linear rails (two or three dependent axes).

This probably isn't much applicable to 3d printers. It is targeted more for cnc mills and routers and machines like pick-and-place.

With a small modification it would be relatively simple to support directional leadscrew wear (different profile coming vs going) as well as backlash. I don't use either.

The request tracks a lengthy/instructive dialog on the forum at: https://forum.duet3d.com/topic/20308/cnc-screw-mapping

Comments

1. The code and doc are at least usable other than they occupy M800, 801, 802, and 810 currently - which I assume is a non-starter. I have done zero performance testing or optimization on the code. I also have slightly different usual code style so it may have style conflicts. 
2. This translates the incoming g-code linear stream hence lines are endpoint-accurate only. At least for me and most applications I can think of that's irrelevant. In the cases where it really matters (for intersections) adding internal nodes to the lines can take care of it.

MZachmann, 2021-01-31:
Here's the doc file explanation of the enhancement. 
[Screw Mapping in RRF.docx](https://github.com/Duet3D/RepRapFirmware/files/5899703/Screw.Mapping.in.RRF.docx)

T3P3, 2024-03-06:
Whis is a fairly advanced feature thats only really useful to people who can correctly determine the maps. please commend to this thread if its something you would use, and preferably link to the machine that you have that you would use it on.

MZachmann, 2024-03-06:
I use this on my CNC, where it's necessary, all the time but I agree this is an advanced feature. It's better than, but roughly equivalent to screw mapping in mach4 or screw compensation in path pilot. For a 3d printer this has no real value since dimensions aren't critical. For a milling machine or CNC router it does have applications and is moderately easy to calculate. On my CNC it increases accuracy from 1/32" to .003". There's a github writeup in my fork of this: https://github.com/MZachmann/RepRapFirmware and the source code has a good intro comment. The code introduces virtually no delay unless it's being used and the use cost is minimal - but there is a moderate amount of code and it won't fit in an older board.

T3P3, 2025-02-07:
@MZachmann We are nearly at the point of releasing 3.6 and starting to look at what to work on for 3.7. There has been a large number of changes between 3.4.x and 3.6, especially in the motion system. Would it be possible to update you frok to the 3.6-dev codebase and test against that?

MZachmann, 2025-02-11:
Sure. I'll update this in a day or two.

MZachmann, 2025-02-25:
I'm pretty sure I've got a working update. I need a replacement for the AdjustPosition I was doing but that's relatively minor.