Duet3D/RepRapFirmware #472: Duet 3 MB6HC -  In-Application upgrade from 3.1.1 to 3.2 failing

Ralms, 2021-02-05:
Hi there, 

I'm running my Duet 3 MB6HC in standalone mode. 
To update it, I've uploaded the `Duet2and3Firmware-3.2.zip` file to System using the Web interface however it failed with error:

` Error: M997: In-application programming binary "Duet3_SDiap_MB6HC.bin" not found`.

The Duet Web Control did update to 3.2.0 from 3.1.1, however the Duet Firmware remained in `RepRapFirmware for Duet 3 MB6HC 3.1.1 (2020-05-19b2)`

On the 3.1.1 release, there is a `Duet3_SDiap_MB6HC.bin` file, however in the 3.2 there isn't, only a `Duet3_SDiap32_MB6HC.bin`.

Is the SDiap different from the SDiap32? 
How can I have the Duet upgraded to the latest version considering this error? 

Thank you.

dc42, 2021-02-06:
Please ask support questions on the forum, https://forum.duet3d.com/.

closed 2021-02-06 by dc42

Ralms, 2021-02-06:
I'm fairly disappointed with this attitude from Duet.
You change something, don't provide any warnings or instructions and they frame it as it's my issue.
Maybe I shouldn't have bought my Duet 3.

dc42, 2021-02-06:
I'm, sorry, but you need to distinguish between Github issues and a support forum. They are NOT the same.

Ralms, 2021-02-06:
Exactly, totally agree.
And this is the exact reason I've raised this issue here, because so far everything indicates to an issue with the 3.2 release even if a simple file name change. 
I've not been the first facing this issue and clearly won't be the last. 
But it's easier to blame it on user error.

dc42, 2021-02-07:
The issue is that you did not fully install firmware 3.1.1.

Ralms, 2021-02-08:
> The issue is that firmware 3.1.1 was not fully installed on your board. The missing file is from the 3.1.1 release.

The 3.1.1 firmware was fully installed on my board, but I clean the System folder by deleting any .bin file or .gz file after the update, otherwise is very messy.

I've uploaded the 3.1.1 files back to System, then uploaded 3.2.2 and the update worked. 

But to be honest, maybe an improvement for the future, where firmware files (.bin, etc) are separated from configurations (.g config files) or a toggle to hide them. 
This need to keep the bin files from a previous update on the system folder is a bit cumbersome.

dc42, 2021-02-09:
To upgrade (or downgrade) firmware, the firmware that is currently running has to load a compatible IAP (in-application-programmer) binary file into RAM, then transfer control to that binary, which then erases the flash memory and loads the new firmware. So you need to have the IAP binary corresponding to the currently-running firmware version in order to upgrade or downgrade firmware via the web interface. It doesn't matter if you have IAP files corresponding to other firmware versions on the SD card too, because whenever we make an incompatible change to the IAP file, we change the filename. I hope this clarifies the situation.

We have plans to move firmware files into a separate /bin folder on the SD card.

Ralms, 2021-02-09:
> To upgrade (or downgrade) firmware, the firmware that is currently running has to load a compatible IAP (in-application-programmer) binary file into RAM, then transfer control to that binary, which [...]

It does and I appreciate the explanation. 
Thank you.