Duet3D/RepRapFirmware #473: Print Object's height and in conjunction layer count is corrupted by G1 move inside Ending Script.

ryan99alero, 2021-02-07:
I use Simplify3D and IdeaMaker to convert my Fusion360 cad files into Gcode files.  I've always had Starting and Ending scripts in my slicer's.  Everything has been fine for awhile until I recently went to Version 3 of RepRap.  I can't say what version I was coming from and going to.  I believe I was on a beta version of 3 that didn't do this.  If my Ending Script moves the Head home for xy and z whatever value is in Z is what value gets displayed in the GUI.  Plus when you go to run the Job it then correlates that new reported height into the number of layers based on your layer height.  

I discovered this while trying to troubleshoot why my Bed heater was stopping to heat.  I'd even do a reset and try changing the temp value prior to the system error out because of the spread between target and desired temp went outside the allow spread.  Below is what my ending script is.  I attached a screenshot to show the same file uploaded one with and one without a Z height on my G1 command below.  Only difference was I removed just the Z5.

M106 S0		;Fan Off
M104 S0 T1		;Turn Off Head1
M104 S0 T0		;Turn Off Head2
M140 S0		;Turn Off Bed Heater
M117 "Print Complete."	;Inform Web Print Done
G1 X1 Y1 Z5 F12000	;Move X and Y back Home
G4 S100  		;Wait 100 seconds and turn off fans
M106 P1 S0		;Disabled Fans
M106 P2 S0		;Disabled Fans
M84		;Stop Idle Hold

[image]
[config.g.zip](https://github.com/Duet3D/RepRapFirmware/files/5938225/config.g.zip)

ryan99alero, 2021-02-08:
So I went back and tried an older job that printed just fine in the past.  It also failed on printing.  Whats odd is I know its not a heater issue as I an not print and set the bed temp to any temp and it'll hold that temp for hours with no issue.  I've tried putting it at 75 for 6 hours and 120c for 3 hours.  Held temp perfect.  Just when printing and after a short amount of time it fails with statement "Printing paused. Heater fault".  Going to try a few more things before I revert firmware to an on older version to see if that allows me to print.

T3P3, 2023-05-04:
please discuss on the forum if this is still an issue in RRF 3.4.5

closed 2023-05-04 by T3P3

dc42, 2023-05-05:
This issue (end GCode commands that increase Z causing incorrect deduction of object height) and the workaround for it have been discussed many times on the forum. We should add the workaround to the troubleshooting FAQ.