Duet3D/RepRapFirmware #475: Hotend PID tuning stuck at equilibrium temperature

tetele, 2021-02-11:
Using RRF 3.2. I tried to tune my main print head using `M303 T0 S260`.

As it turns out, when placed close to the bed and with the part cooling fan at 100%, my hotend can only reach ~240 degrees. I did it in this configuration because of some advice I received that you should tune the hotend in approximately the printing position, although I'm not sure that's accurate.

As a result, after having reached the requested 260 degrees, after settling, in the "fan on" phase, the hotend reached a steady ~238 degrees and just stays there for about half an hour, without any way for me to cancel the tuning except for resetting the board.
[image]

I would have expected the algorithm to realize the fact that it can't reach the requested temperature with the fan on, just as it does when the fan is off.

dc42, 2021-02-12:
Thanks, it looks like we should introduce a timeout for heating back up to temperature in that phase.

mydevpeeps, 2021-02-12:
Could we have it recognize an already defined M570 code?

T3P3, 2026-08-24:
Generalising the requirement for a timeout allows us to close the issue: Configurable timeout for M303 tuning #948
as a duplicate.
> M303 autotuning has a 30 minute time limit for bed/chamber to reach temperature, and a 7 minute time limit for hot end heaters to reach temperature. As machines get larger, this may not be long enough.

>Add a parameter to M303 to override the default timeout.

T3P3, 2026-08-24:
when working on heater tuning consider implementing #1010 Add M303 cancelling to M108

T3P3, 2026-08-24:
When working on heater tuning, consider adding the warning that it will not function correctly when the `I1` option is used in `M307` from #570 , until #570 is implemented

T3P3, 2026-09-16:
Forum threads:
https://forum.duet3d.com/topic/34758
https://forum.duet3d.com/topic/34027
https://forum.duet3d.com/topic/33073