Duet3D/RepRapFirmware #479: Compilation error "template argument 1 is invalid" in "type_traits" header

M-Pixel, 2021-02-21:
```
[286/340] Building CXX object CoreN2G/CMakeFiles/CoreN2G.dir/src/Stream.cpp.obj
FAILED: CoreN2G/CMakeFiles/CoreN2G.dir/src/Stream.cpp.obj 
/usr/bin/arm-none-eabi-g++  -DDUET3MINI_V04 -DHAS_LINUX_INTERFACE -DRTOS -DSUPPORT_CAN=1 -DSUPPORT_SDHC=1 -DSUPPORT_USB=1 -DUSE_CAN0 -D__SAME54P20A__ -Dnoexcept="" -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/SAME5x_C21 -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/SAME5x_C21/SAME5x -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/SAME5x_C21/SAME5x/hal/include -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/SAME5x_C21/SAME5x/hal/utils/include -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/SAME5x_C21/SAME5x/hri -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/SAME5x_C21/SAME5x/Config -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/SAME5x_C21/SAME5x/usb -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/SAME5x_C21/SAME5x/usb/device -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/SAME5x_C21/SAME5x/usb/class/cdc -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/atmel/SAME54_DFP/1.1.134/include -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/arm/CMSIS/5.4.0/CMSIS/Core/Include -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/../RRFLibraries/src -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/../CANlib/src -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/../FreeRTOS/src/include -I/mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/../FreeRTOS/src/portable/GCC/ARM_CM4F -mthumb -O2 -g -DNDEBUG   -c -ffunction-sections -fdata-sections -Wundef -Wdouble-promotion -fsingle-precision-constant -Wall -O2 -nostdlib -fno-threadsafe-statics -fno-rtti -DRTOS -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-exceptions -std=gnu++17 -MD -MT CoreN2G/CMakeFiles/CoreN2G.dir/src/Stream.cpp.obj -MF CoreN2G/CMakeFiles/CoreN2G.dir/src/Stream.cpp.obj.d -o CoreN2G/CMakeFiles/CoreN2G.dir/src/Stream.cpp.obj -c /mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/Stream.cpp
In file included from /usr/arm-none-eabi/include/c++/8.2.1/bits/move.h:55,
                 from /usr/arm-none-eabi/include/c++/8.2.1/bits/stl_pair.h:59,
                 from /usr/arm-none-eabi/include/c++/8.2.1/bits/stl_algobase.h:64,
                 from /usr/arm-none-eabi/include/c++/8.2.1/bits/specfun.h:45,
                 from /usr/arm-none-eabi/include/c++/8.2.1/cmath:1892,
                 from /mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/../RRFLibraries/src/General/SimpleMath.h:12,
                 from /mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/CoreIO.h:19,
                 from /mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/Print.h:23,
                 from /mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/Stream.h:8,
                 from /mnt/c/Users/Max/Code/RepRapBuildSuite/CoreN2G/src/Stream.cpp:9:
/usr/arm-none-eabi/include/c++/8.2.1/type_traits:447:65: error: template argument 1 is invalid
     struct is_function<_Res(_ArgTypes...) _GLIBCXX_NOEXCEPT_QUAL>
                                                                 ^
```

Searching for similar issues around the internet, most cases were resolved by updating the version of GCC that was used.

I installed this version of gcc-arm-none-eabi, as indicated in the instructions:

> To build firmware version 2.03beta3 and later use version 2018-q4-major

Am I supposed to be using a newer version for firmware version 3.2?

It appears to me that gcc-arm-none-eabi does _not_ source any system-provided header files, so it seems unlikely that it would be an incompatibility with the host system, but I won't rule it out entirely at this point.  I'm using Ubuntu 20.04.

dc42, 2021-02-21:
See the RRF github wiki for a version of the build instructions file that covers more RRF versions.

M-Pixel, 2021-02-22:
It turns out that the issue was that I mistakenly included the `noexcept=` definition in C++ compilation.  It should only be defined as such for C compilation.

closed 2021-02-22 by M-Pixel