Duet3D/RepRapFirmware #485: add support for the M150 command to the Duet 2 Maestro

cyrax72, 2021-03-06:
good day. It would be very nice to add support for the M150 command to the Duet 2 Maestro and other boards .. I understand that there is no ready-made connector with level matching for the DotStar and Neopixel tapes, but the matching is easy to do by yourself. This is sorely lacking, and changing the board to version 3 just because of the tape is very expensive. Well, at the same time I would like to say that RRF is the best firmware in the world!

dc42, 2022-03-29:
Which pin on the Maestro do you suggest we use?

cyrax72, 2022-03-30:
Thanks for your reply. I do not have a screen, so all pins of connectors 12864_exp1 and 12864_exp2 are free. Or for example EXP_0 or EXP_1

T3P3, 2023-05-04:
https://docs.duet3d.com/en/User_manual/Reference/Gcodes#m150-set-led-colours

This should now be possible:
In older firmware versions this command is only supported on controllers that have an output connector for DotStar or NeoPixel LEDs. More recent firmware's remove this limitation.

Note in 3.5.0beta4 and later the LED strip needs to be defined with M950:
https://docs.duet3d.com/en/User_manual/Reference/Gcodes#m950-create-heater-fan-spindle-or-gpioservo-pin

closed 2023-05-04 by T3P3