Duet3D/RepRapFirmware #490: M585 does Not Update laststopheight

wildblue, 2021-03-15:
When using m585 the field LastStopHeight does Not get updated

dc42, 2021-03-16:
This needs some thought, because the stop position is only a height if probing was in the Z direction. Also, as the defined function of M585 is to set the tool offset, the stop height is in essence defined by the M585 axis parameter, and the tool offset is adjusted to give that stop height.

wildblue, 2021-03-16:
With conditional gcode we are now able to iterate over single commands like M585. If the result of a single probing action (this is also valid for G30; consider multiple probing with one command as well) is Not accessible via OM, the whole Idea behind this awesome approach would get a hard hit.

dc42, 2021-03-16:
I think you can already work out the M585 stop height from the new tool offset.

wildblue, 2021-03-16:
Regardless of what M585 can be used for, it is using a probe which has a set of variables describing its status which includes also the laststopheight. What is happening to M585's variables is secondary here.

For what does the variable laststopheight else exists? Get rid of it, if it is not updated by a sensor probe.
Or maybe even better: just implement the missing code for M585? You decide!

T3P3, 2023-05-04:
Please open a discussion on the forum for this if it does not already exist (if it does link it from there).

T3P3, 2024-03-06:
please reopen if still relevant

closed 2024-03-06 by T3P3