Duet3D/RepRapFirmware #495: When probing with M558, the parameter "A" is not working

wildblue, 2021-03-19:
The command M558 has an option (A) to specify how many times an individual point should be probed. This Parameter is currently Not working.

3.3b2, duet2wifi, coreXY

T3P3, 2023-05-04:
@wildblue please can you see if this is still the case with RRF 3.4.5

AndyEveritt, 2023-06-06:
Replicated in 3.5.0-b3.
Also exhibited a weird behaviour that the tool was unselected (and `tfree` called) before probing with `M585`.
`G30` does probe multiple times

AndyEveritt, 2023-06-06:
Behaviour seems consistent in 3.4.5

T3P3, 2024-05-20:
@AndyEveritt is this still an issue in 3.5.1?

T3P3, 2025-05-07:
The A parameter for M558 only applies to G29 and G30 moves, not currently to M585 or G38.x.

Changed to an enhancement and tentatively allocated as a feature request to 3.7

T3P3, 2026-09-16:
Forum thread: https://forum.duet3d.com/topic/34077