Duet3D/RepRapFirmware #498: M80 not working went sent from Gcode file

broncosis, 2021-03-21:
I dove on the new beta for the variables to make start code more uniform think I may have found a bug 

But as I use ATX power switching I noticed M80 is not launching from the job file or the Macro file 
M81 apeears to be working from macro files 
and I can send M80 on the console and it works as well at the button on the dash board

broncosis, 2021-03-21:
Running on a duet2 wifi    

[config.txt](https://github.com/Duet3D/RepRapFirmware/files/6176969/config.txt)
attached is my config

broncosis, 2021-03-21:
OK seems to be fine called from a Macro run from the webUI but not when the Macro is called from a job file
also works from the Job File 
I can work around this but now but still worth noting

dc42, 2022-03-29:
Please re-test using RRF 3.4.0.

broncosis, 2022-04-27:
I just updated sorry for the slow response I will try this thanks

closed 2023-05-04 by T3P3