Duet3D/RepRapFirmware #505: Duet Maestro Data Corruption where Tool count > 18

nanoplane, 2021-04-05:
When I create more than 18 tools, It appears I get data corruption since the hardware acts strangely (reverse Extrusion on some drives).  In addition, DWC goes wonky, continuous connection failures (service not available), and all Tools listed are Tool 0 with no sensor data available.

RRF Version 3.2.2, DWC Version 3.2.2

DWC screen shot of tool area.
[image]

dc42, 2021-04-06:
To look at this further, I need at least a M122 report after this has happened.

nanoplane, 2021-04-06:
Here you go:
M122
=== Diagnostics ===
RepRapFirmware for Duet 2 Maestro version 3.2.2 running on Duet Maestro 1.0
Board ID: 08DLM-9T6R1-MD3T0-6JKD0-3SD6T-9FRTQ
Used output buffers: 3 of 24 (24 max)
=== RTOS ===
Static ram: 22284
Dynamic ram: 75804 of which 236 recycled
Never used RAM 16508, free system stack 150 words
Tasks: NETWORK(ready,190) HEAT(blocked,333) TMC(blocked,111) MAIN(running,455) IDLE(ready,21)
Owned mutexes: W5500(NETWORK)
=== Platform ===
Last reset 34:31:42 ago, cause: software
Last software reset at 2021-04-04 20:51, reason: User, GCodes spinning, available RAM 18808, slot 1
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x00000000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
Error status: 0x04
Aux0 errors 0,0,0
MCU temperature: min 30.3, current 30.9, max 31.2
Supply voltage: min 23.8, current 23.9, max 23.9, under voltage events: 0, over voltage events: 0, power good: yes
Driver 0: position 0, standstill, read errors 0, write errors 0, ifcnt 60, reads 15662, writes 0, timeouts 0, DMA errors 0
Driver 1: position 0, standstill, read errors 0, write errors 0, ifcnt 60, reads 15662, writes 0, timeouts 0, DMA errors 0
Driver 2: position 2000, ok, read errors 0, write errors 0, ifcnt 60, reads 15662, writes 0, timeouts 0, DMA errors 0
Driver 3: position 0, standstill, read errors 0, write errors 0, ifcnt 44, reads 15661, writes 0, timeouts 0, DMA errors 0
Driver 4: position 0, standstill, read errors 0, write errors 0, ifcnt 44, reads 15661, writes 0, timeouts 0, DMA errors 0
Driver 5: position 0, standstill, read errors 0, write errors 0, ifcnt 44, reads 15662, writes 0, timeouts 0, DMA errors 0
Driver 6: position 0, standstill, read errors 0, write errors 0, ifcnt 44, reads 15662, writes 0, timeouts 0, DMA errors 0
Date/time: 2021-04-06 07:23:43
Slowest loop: 12.82ms; fastest: 0.11ms
I2C nak errors 0, send timeouts 0, receive timeouts 0, finishTimeouts 0, resets 0
=== Storage ===
Free file entries: 8
SD card 0 detected, interface speed: 15.0MBytes/sec
SD card longest read time 2.1ms, write time 0.0ms, max retries 0
=== Move ===
DMs created 83, maxWait 2ms, bed compensation in use: none, comp offset 0.000
=== MainDDARing ===
Scheduled moves 8, completed moves 7, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state 3
=== AuxDDARing ===
Scheduled moves 0, completed moves 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== Heat ===
Bed heaters = 0 -1, chamberHeaters = -1 -1
Heater 0 is on, I-accum = 0.2
=== GCodes ===
Segments left: 0
Movement lock held by File
HTTP is doing "M25" in state(s) 0
Telnet is idle in state(s) 0
File is idle in state(s) 1 8, running macro
USB is idle in state(s) 0
Aux is idle in state(s) 0
Trigger is idle in state(s) 0
Queue is idle in state(s) 0
LCD is idle in state(s) 0
Daemon is idle in state(s) 0
Autopause is idle in state(s) 0
Code queue is empty.
=== Network ===
Slowest loop: 35.42ms; fastest: 0.06ms
Responder states: HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0), 0 sessions
HTTP sessions: 2 of 8
Interface state active, link 100Mbps full duplex

nanoplane, 2021-04-06:
Here's a sample file that created this
[tooltest.gcode.txt](https://github.com/Duet3D/RepRapFirmware/files/6265375/tooltest.gcode.txt)

dc42, 2021-04-07:
This is caused by RRF running out of message buffers, due to the very long response needed when DWC requests the 'tools' subtree of the object model. It is best fixed in DWC.

closed 2021-04-07 by dc42