Duet3D/RepRapFirmware #508: Print finished after pause from sd card

nikscha, 2021-04-24:
In my gcode file there's a M226 command to pause the printer. After I resume the print, the printer keeps printing but on DWC the print is marked as 100% and I get presented with the "print again" button.

at 1:33 the print was paused due to the M226 command in the gcode file.
I inserted a nut into the print and resumed the print (worked just fine).
But when I checked DWC, I saw the usual "print is done" screen.
I issued a M25 to pause the print via the panelDue but got the error below. 

[image]

Reproducing the issue via DWC didn't work.

This is my setup:
Board: Duet 3 Mini 5+ and raspi 4 
DSF Version: 3.2.2
DWC v3.2.2
Firmware: RepRapFirmware for Duet 3 Mini 5+ 3.2.2 (2021-02-11)

jbjhjm, 2021-10-07:
Experiencing this too. Still present in 3.4.0b3.
It may be related to the pause issue mentioned here: https://github.com/Duet3D/RepRapFirmware/wiki/Changelog-RRF-3.x-Beta-&-RC#reprapfirmware-340beta4

When the print gcode calls M226, pause mode will be started, but when calling resume, it throws an error (stating there is a gcode parse error).

When the print gcode uses M25, pause mode will start too and DWC will reflect pause state correctly too. Though when clicking resume, the print will continue and immediately pause itself again. I suspect that execution stops without setting the current action index to the next action, so it can never continue past that M25 call.

dc42, 2021-10-13:
Thanks, this is fixed in RRF/DSF 3.4.0beta5.

closed 2022-03-29 by dc42