Duet3D/RepRapFirmware #513: Bug using G18/G19 with release 3.3 beta3

derd3, 2021-05-01:
Hi

First, thanks for enabling the use of G18 and G19 commands!
I started testing with release 3.3 beta3 on a Duet 2 Wifi in cnc-mode and stumbled across three following issues:

1- RepRap seems to be not supporting the K parameter for G2/3 which would be used to give offsets in Z. Just like it ist done with I and J. This can be circumvented by using G2/3 with the R Parameter, at least if the post processor supports it. 

2- If I run a job where a G18/19 command is used for a G2/3 command like "G18 G2 X0 Y0 ...."
the next job takes over the plane for its arc-moves that was set in the previous job. I think after a job has ended or is canceled all moves should again be made in the XY plane by default without the need for explicitly setting it.

3- The postprocessor I use in Autodesk CAM is the HAAS - A-axis(preNGC)... which generates G-code like this:
G1 X0 Y0
X1
Y1
which let the machine end up at X1Y1 and normally runs without issues

If an arc move in a different plane is made using G18/19 this does not work anymore. The code only runs correctly if every line gets its own G-command: 
**G18** G2 Y11.57 Z-6.685 R35.643
G1 Y11.584 Z-6.687
G1 Y11.598 Z-6.683
Y11.615 Z-6.679 (this line would not be executed)

Funny enough though
**G17** G2 Y11.57 Z-6.685 R35.643
G1 Y11.584 Z-6.687
Y11.598 Z-6.683
works just fine.

dc42, 2021-05-01:
1 - I confirm this is a bug. I have implemented a fix in the source code ready for 3.3 stable release. Are you able to test it for me, and if so, which firmware build do you need?
2 - Please raise a separate issue (feature request) for this.
3 - Please re-test this. I cannot see any mechanism whereby the current selected plan could cause implicit G0/G1 commands to fail.

dc42, 2021-05-03:
Re #1 I have implemented a provisional fix. New binaries are available at https://www.dropbox.com/sh/dlb58vkmu1u4fkx/AAAelkXSfRKVwI6_yqRnhGHPa?dl=0.

derd3, 2021-05-03:
2- will do so
1- Would be a pleasure, but I wanted to verify the findings on 3- first 
3- I can absolutely confirm my observations but I now also think it has nothing to do with the selected plane.
The exact G-code used for the tests is as followed:
G90
G4 S2

G0 Z0 Y0 X0 
G1 X50 F2000
Y50
X0
Y0

G3 X50 Y50 R50
X0 Y0 R50
G1 X50
Y50
X0
Y0

3.2.2: works as expected.
3.3 beta3: implicit execution does not work or only kind of for G1. Implicit execution does work for G3.
3.3 beta3 (the one from the Dropbox folder):  implicit execution does not work for neither G1 nor G3
3.3 RC1: same as for 3.3 beta3

See the picture for more details
[image]
Numbers give the order the moves were made.

I performed all test like this - yes repeteately :)
1. upload new Duet2combindedFirmware.bin => install update
2. Home all axis from the dashboard
3. Run the same file containing the code stated above.
Since I assume a restart is enough I did not power cycle the board in between the tests.

Changing the G3 to use I and J parameters does not change the observed behaviour.
The HW used: Duet2 Wifi, controled via the web interface from a raspberry pi4 2Gb

derd3, 2021-05-03:
Using the FW 3.3 RC1 downloaded from Github I tested the G18/19 commands anyway.

The K parameter has no effect. Instead one needs to use I and J for the offsets. So for an arc in the xy plane the offset of the rotation point in X is given by the I parameter => correct, But for a Z offset the J parameter must be used instead of K.
for example: G18 G3 X50 Y0 I25 K-20 would need to be written as G18 G2 X50 Y0 I25 J-20 to get the desired result.

For an arc in the YZ plane using G19, to get the desired result the offset in global Y must be given by the parameter I instead of J, the offset in Z by J instead of K.

Also for both arcs in G18 and G19 the G2 and G3 commands need to be flipped to get the desired movement - at least comapred to the backplot of my Autodesk CAM.

As far as I understand the offsets I J K are to be given in the global coordinate system and not in a local one. The direction of rotation is also determined by the global coordinate system. G3 being a rotation in the mathematically positive direction on the corresponding xz or yz plane.

dc42, 2021-05-04:
Please test G18/19 with G2/3 using the 3.3RC1+1 files I linked to in my previous reply. Note, this version does not support implicit G2/G3 because that is not part of the NIST standard.

derd3, 2021-05-06:
Arc moves in G19 were successful.
Arc moves in G18 needed flipped G2 and G3 to go the right way round.

dc42, 2021-05-07:
Thanks. G2/G3 direction after G18 is fixed in 3.3RC2.

closed 2021-05-07 by dc42