Duet3D/RepRapFirmware #515: M593 DAA not working at all.

64bittuning, 2021-05-05:
Recently updated to RRF3.3b3 from 2.05.1.

Re-ran all DAA tests and prints. Testing is repeatable at 30.5-31.3
Set M201 to 2500, 7000 and 10000 for X and Y axis, no difference.
Set M566 as low as 120 on both X and Y axis, no difference.

When commanding in console
M593 F31.3, it returns:
5/5/2021, 12:23:28 PM | M593 F31.3

When commanding in console: 
M593 F31.3 P6, it returns
M593 F31.3 P6   Error: M593: parameter 'P' too high

Sending M593 P6 by itself returns 
M593 P6   Error: M593: parameter 'P' too high

M593 Pn will return an acceptable response with 0 through 4. 5 and 6 will both echo M593: parameter 'P' too high
[image]

dc42, 2021-05-05:
3.3beta3 has been superseded by 3.3RC1.

64bittuning, 2021-05-05:
Testing now. Will report back.

64bittuning, 2021-05-05:
M593 P5 and P6 values still report back with **Error: M593: parameter 'P' too high**

M593 Fn with no P value now produces a significantly improved print.

dc42, 2022-03-29:
M593 has been completely reworked in RRF 3.4.0. DAA is no longer supported because the new forms of input shaping supported in 3.4.0 are better.

closed 2022-03-29 by dc42