Duet3D/RepRapFirmware #521: FR wifi esp support for duet3 6hc

Lino77, 2021-05-13:
if a rpi is running on the duet3 6hc why is the esp not supported, i have looked at the circuit diagrams the board actually provides everything that hardware is moderately required for it,

dc42, 2021-05-14:
The MB6HC does not have a ESP module.

Lino77, 2021-05-14:
I know that he doesn't have one. but you could connect one to the Rpi port and take RX TX from IO_0 or IO_1 and from the other IO ports, or I think wrong

romalb47, 2021-09-07:
Or use an rpi as wifi controller, it's not really more costly and add a lot's of advantage. Like a VPN or tactil screen for administration. Adding an ESP to the board is redundant and add code that load the MCU and must be supported.