Duet3D/RepRapFirmware #522: How to get real coordinates while driving?

selenur, 2021-05-15:
How to get real coordinates while driving?
If coordinates XY (0:0) and send the command **G1 F10 X100**, then the next command **M114** show the coordinates XY as (100:0), but the real coordinates of the x-axis are not yet 100.

PatrikRindlisbacher, 2022-05-30:
I discovered the same problem.

With CNC milling, this is elementary.

The display of the coordinates should always be up to date.

T3P3, 2022-05-30:
you can segment moves and then the co-ordinates will be shown in the segments.

See:
https://docs.duet3d.com/User_manual/Reference/Gcodes#m669-set-kinematics-type-and-kinematics-parameters
Specifically the S&T parameters

please use our forum (forum.duet3d.com) for support.

closed 2022-05-30 by T3P3

dc42, 2022-05-30:
There are two different sets of coordinates: user coordinates and machine coordinates. When the machine is moving, the machine coordinates are kept more up to date than the user coordinates.

PatrikRindlisbacher, 2022-05-30:
Thanks for the quick reactions.

When CNC milling from CAM, it is **important** to always see the **current** user coordinates G54-G59.

Unfortunately, segmentation is **not** a solution.