Duet3D/RepRapFirmware #523: M557 error when used with variables

nmsmith22389, 2021-05-17:
# System

<table>
	<tr align="left">
		<th>
			Board
		</th>
		<td>
			Duet 3 Mini 5+ <em>(Mini5plus)</em>
		</td>
	</tr>
	<tr align="left">
		<th>
			DSF Version
		</th>
		<td>
			3.3-rc2
		</td>
	</tr>
	<tr align="left">
		<th>
			Firmware
		</th>
		<td>
			RepRapFirmware for Duet 3 Mini 5+ 3.3RC2 <em>(2021-05-11)</em>
		</td>
</table>

# Description

Code `M557` throws an error when used with any variables as parameters.

For example, when using something like:

```gcode
M557 X{var.xMin}:{var.xMax} Y{var.yMin}:{var.yMax} P8
```

the firmware returns:

```
Error: M557: Wrong number of values after '''X''', expected 2
```

I've tried all sorts of combinations in an attempt to debug it but it seems that the presence of even one variable/interpolation causes the error.

# Diagnostics

<details>
	<summary><code>M122 B0</code></summary>
	<p>
		<pre><code>
=== Diagnostics ===
RepRapFirmware for Duet 3 Mini 5+ version 3.3RC2 (2021-05-11 14:55:40) running on Duet 3 Mini5plus WiFi (SBC mode)
Board ID: 524WM-Y296U-D65J0-40KM8-MB03Z-HJSAK
Used output buffers: 1 of 40 (20 max)
=== RTOS ===
Static ram: 102476
Dynamic ram: 95324 of which 0 recycled
Never used RAM 43048, free system stack 96 words
Tasks: SBC(ready,4.5%,296) HEAT(delaying,0.0%,318) Move(notifyWait,0.1%,194) CanReceiv(notifyWait,0.0%,773) CanSender(notifyWait,0.0%,363) CanClock(delaying,0.0%,339) TMC(notifyWait,0.7%,106) MAIN(running,93.9%,307) IDLE(ready,0.0%,29) AIN(delaying,0.8%,264), total 100.0%
Owned mutexes: HTTP(MAIN)
=== Platform ===
Last reset 11:02:28 ago, cause: software
Last software reset at 2021-05-17 01:15, reason: User, GCodes spinning, available RAM 43824, slot 1
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x00000000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
Error status: 0x00
Aux0 errors 0,0,0
Aux1 errors 0,0,0
MCU revision 3, ADC conversions started 39748219, completed 39748219, timed out 0, errs 0
Step timer max interval 1487
MCU temperature: min 31.6, current 31.8, max 31.8
Supply voltage: min 24.1, current 24.2, max 24.2, under voltage events: 0, over voltage events: 0, power good: yes
Heap OK, handles allocated/used 99/12, heap memory allocated/used/recyclable 2048/566/328, gc cycles 0
Driver 0: position 21600, standstill, SG min/max 88/88, read errors 0, write errors 0, ifcnt 46, reads 1567, writes 0, timeouts 0, DMA errors 0
Driver 1: position 11600, standstill, SG min/max 110/110, read errors 0, write errors 0, ifcnt 50, reads 1567, writes 0, timeouts 0, DMA errors 0
Driver 2: position 45708, standstill, SG min/max 0/0, read errors 0, write errors 0, ifcnt 49, reads 1568, writes 0, timeouts 0, DMA errors 0
Driver 3: position 0, standstill, SG min/max 2/2, read errors 0, write errors 0, ifcnt 49, reads 1568, writes 0, timeouts 0, DMA errors 0
Driver 4: position 0, standstill, SG min/max 0/0, read errors 0, write errors 0, ifcnt 20, reads 1567, writes 0, timeouts 0, DMA errors 0
Driver 5: position 0, assumed not present
Driver 6: position 0, assumed not present
Date/time: 2021-05-17 12:17:58
Cache data hit count 4294967295
Slowest loop: 1.21ms; fastest: 0.08ms
=== Storage ===
Free file entries: 10
SD card 0 not detected, interface speed: 0.0MBytes/sec
SD card longest read time 0.0ms, write time 0.0ms, max retries 0
=== Move ===
DMs created 83, maxWait 0ms, bed compensation in use: none, comp offset 0.000
=== MainDDARing ===
Scheduled moves 145142, completed moves 145142, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== AuxDDARing ===
Scheduled moves 0, completed moves 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== Heat ===
Bed heaters = 0 -1, chamberHeaters = -1 -1
=== GCodes ===
Segments left: 0
Movement lock held by null
HTTP* is doing "M122 B0" in state(s) 0
Telnet is idle in state(s) 0
File* is idle in state(s) 0
USB is idle in state(s) 0
Aux is idle in state(s) 0
Trigger* is idle in state(s) 0
Queue* is idle in state(s) 0
LCD is idle in state(s) 0
SBC is idle in state(s) 0
Daemon is idle in state(s) 0
Aux2 is idle in state(s) 0
Autopause is idle in state(s) 0
Code queue is empty.
=== CAN ===
Messages queued 268, send timeouts 0, received 357, lost 0, longest wait 0ms for reply type 0, peak Tx sync delay 276, free buffers 17 (min 17)
=== SBC interface ===
State: 4, failed transfers: 0
Last transfer: 3ms ago
RX/TX seq numbers: 19311/2301
SPI underruns 0, overruns 0
Number of disconnects: 1, IAP RAM available 0x106f0
Buffer RX/TX: 0/0-0
=== Duet Control Server ===
Duet Control Server v3.3-rc2
Code buffer space: 4096
Configured SPI speed: 8000000 Hz
Full transfers per second: 34.74
Codes per second: 0.03
Maximum length of RX/TX data transfers: 5584/1664
		</code></pre>
	</p>
</details>

<details>
	<summary><code>M122 B121</code> <em>Toolboard1LC</em></summary>
	<p>
		<pre><code>
Diagnostics for board 121:
Duet TOOL1LC firmware version 3.3RC2 (2021-05-11 14:59:50)
Bootloader ID: SAMC21 bootloader version 2.3 (2021-01-26b1)
Never used RAM 4056, free system stack 0 words
Tasks: Move(notifyWait,1.1%,99) HEAT(delaying,1.4%,81) CanAsync(notifyWait,0.0%,62) CanRecv(notifyWait,0.2%,77) CanClock(notifyWait,0.1%,66) TMC(delaying,21.1%,57) MAIN(running,39.5%,339) IDLE(ready,0.0%,40) AIN(delaying,36.5%,62), total 100.0%
Last reset 11:02:46 ago, cause: software
Last software reset at 2021-05-16 05:00, reason: AssertionFailed, available RAM 4056, slot 0
Software reset code 0x0120 ICSR 0x00000000 SP 0x20003cc4 Task �: Freestk 136 ok
Stack: 00000544 00023174 00019cb5 20003044 00016e33 20003044 00016505 200071c8 00000001 00000001 0000828d 200071c8 200071c8 00000000 20000f50 00000020 00011311 200071c8 00008611 200071d8 007d2230 007d2230 41f12eb0 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5
Driver 0: position 6745314, 400.0 steps/mm, standstill, SG min/max 0/368, read errors 0, write errors 1, ifcnt 56, reads 1496, writes 22, timeouts 0, DMA errors 0, steps req 13560740 done 13560740
Moves scheduled 126159, completed 126159, in progress 0, hiccups 0, step errors 0, maxPrep 657, maxOverdue 12, maxInc 8, mcErrs 0, gcmErrs 0
Peak sync jitter 9, peak Rx sync delay 229, resyncs 48, no step interrupt scheduled
VIN: 24.1V
MCU temperature: min 36.1C, current 37.7C, max 45.0C
Ticks since heat task active 96, ADC conversions started 39714776, completed 39714775, timed out 0, errs 0
Last sensors broadcast 0x00000002 found 1 101 ticks ago, loop time 0
CAN messages queued 477576, send timeouts 0, received 484401, lost 0, free buffers 36, min 35, error reg 110000
dup 0, oos 0, bm 0, wbm 0
Accelerometer detected: no
I2C bus errors 4509, naks 4509, other errors 4509
		</code></pre>
	</p>
</details>

dc42, 2021-05-17:
This is a known limitation of DCS+RRF at present. When running in SBC mode, array-valued parameters cannot be constructed from expressions.

T3P3, 2023-05-04:
believed fixed in 3.4.0:

https://github.com/Duet3D/RepRapFirmware/wiki/Changelog-RRF-3.x

> Array expressions are now supported in GCode parameters that expect multiple values, both in standalone and SBC modes using syntax { expression, expression, ... }. For example: M92 E{var.e1,var.e2}

Please retest in 3.4.5 and if it is still not working then reopen this issue