Duet3D/RepRapFirmware #527: Lasermode after update to 33rc3 nw

wildblue, 2021-06-09:
strange behaviour:
9.6.2021, 12:01:57	Connection established
9.6.2021, 12:01:57	Connection interrupted, attempting to reconnect...
HTTP request timed out
9.6.2021, 12:01:39	m80
9.6.2021, 12:01:19	M452 C"duex.gp2" R32 S0 F1024
9.6.2021, 11:59:28	Connected to 192.168.1.50

It reboots after in laser mode and then running g28

I saw different issues with lasermode and gp1, is this the same issue™?

dc42, 2021-06-09:
Yes, the DueX GP pins and fan outputs cannot be used to control the laser. The latest 3.3 build reports an error if you try to configure a laser on those ports.

closed 2021-06-09 by dc42

wildblue, 2021-06-09:
Following works, but why?
M452 C"duex.e3heat" R32 S0 F1024

i have to be able to use the gpios in the future...

Duetwifi2,3.3.rc3

dc42, 2021-06-09:
It's because the DueX fan ports and GPIO ports are driven through the I2C expander, which has far too much latency to be used inside the step ISR where the laser power is controlled. Whereas the DueX heater ports are driven directly from the microcontroller.