Duet3D/RepRapFirmware #529: Feature request: M92 steps per mm definable for dual axis drive

RetroHQ, 2021-06-18:
It does not seem possible to specify steps per mm uniquely to each motor of a multi motor axis drive. When you have a larger system this becomes more relevant -- I currently have a situation where the two screw drives on the Y axis are marginally different steps/mm ending up with about 2mm variance over a meter. This should be easily compensable in software but instead requires accurately matches lead screws currently.

RetroHQ, 2021-06-21:
If anyone can give me some help on how to get started on this, I'll see if I can sort it.

T3P3, 2023-05-04:
In general RRF expects matched motors and drives for an axis at this point. 

Flagged for consideration. If you still require this feature please open a forum thread in the CNC category (I expect its most relevant to leadscrew driven machines) for discussion there.

RetroHQ, 2023-05-04:
Personally I no longer need this as I got around the issue by going to-and-fro with the supplier until I got a set of lead screws which were evenly enough matched for the job. In general though, leads screws are never perfectly matched -- I think out of 6 I managed to find two which were close enough, but still a little out.

For large format machines this really is a useful feature and would still be useful for me now to fine tune accuracy further.

dc42, 2023-05-04:
This would be a major change because it would no longer be possible to generate one set of steps for all motors on that axis. If the leadscrews have inaccurate thread pitch, they are likely to have non-uniform thread pitch too and need compensation for the non-linearity applied (which has been requested in the past).

RetroHQ, 2023-05-04:
Yes, absolutely, multiple linear sections would be preferable. It may be a major change, but it would be very beneficial with lead screw based machines.

I’m not sure how badly linearity suffers with lead screws, but either way, more flexibility is better.

T3P3, 2026-09-16:
Forum threads:
https://forum.duet3d.com/topic/31991
https://forum.duet3d.com/topic/31172