Duet3D/RepRapFirmware #533: M117 Error on parsing gcode file for job

creek52, 2021-06-23:
Updating standalone Duet 2 Wifi from 3.2.2 to 3.3 has caused an error to be generated while parsing job gcode file.
It returns "Error: in GCode file line 3731 column 9: M117: string too long.",  Always column 9!
if you send an M117 command from the console it produces the correct result displaying the complete line.
An M117 command is generated by PrusaSlicer at every layer change.
GCode file worked fine with v3.2.2 but paneldue and webcontrol now gets filled with errors for the duration of the file.

;LAYER_CHANGE
;Z:0.3
;HEIGHT:0.3
;layer 1 
M117 "Layer 1/66 @0.3mm"    <---error occurs

[image]

dc42, 2021-06-23:
Please provide a sample GCode file that demonstrates this problem. Are these errors also reported if you simulate the file?

creek52, 2021-06-25:
GCode from Prusaslicer 2.3.1 that has the problem at the first error 

G92 E0 ; reset extruder to zero 
;LAYER_CHANGE 
;Z:0.3 
;HEIGHT:0.3 
;layer 1 
M117 "Layer 1/66 @0.3mm" 
;M291 P"Layer 1/66 @0.3mm" S0 T8 
G1 Z0.300 F7800.000 
G1 E-2.00000 F1800.000 
G92 E0 
G1 X77.115 Y77.571 F7800.000 
G1 E2.00000 F1800.000 
;TYPE:Skirt 
;WIDTH:0.4 
\ 

Duet Web Control 3.3.0 
RepRapFirmware for Duet 2 WiFi/Ethernet 3.3 (2021-06-15) 
Duet WiFi Server Version: 1.26 

error occurs with every M117 command. 
Yes, same error occurs during simulation too. Does this mean it's in Duet Web Control 3.3.0, not in the firmware? 

Tried another printer 
The same gcode file on another printer that has 3.2.2 works ok. 
Duet Web Control 3.3.0-rc3 
Board: Duet 2 WiFi (2WiFi) 
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 3.2.2 (2021-02-11) 
Duet WiFi Server Version: 1.26 
This one uses release candidate for Web Control, so I'm believing it is not within web control.

dc42, 2021-06-25:
I can't reproduce this. Running that snippet on my Duet WiFi bench system does not produce any error messages and produces the expected M117 popup. Many Duet users use PrusaSlicer and nobody else has reported this. So I think it must be specific to your file. I asked you to provide a file, but you didn't, just a snippet that I had to cut and paste. There may be non-printing characters in the actual file that would not be passed across using copy and paste.

If you wish to take this further, please start a thread on the forum at https://forum.duet3d.com/ and attach an actual print file that causes the problem to your post.

dc42, 2021-07-06:
Another user reported this on the forum, along with a short file to reproduce it. I have located the problem it will be fixed in RRF 3.4.0, also if we do a 3.3.1 release it will be fixed in that. Meanwhile, a workaround is to use M400 immediately before M117.

closed 2022-03-29 by dc42