Duet3D/RepRapFirmware #534: [Bug]:

natthapolvanasrivilai, 2021-07-06:
### Which Duet products are you using?

- [X] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

RepRapFirmware 3.3

### Duet Web Control Version

Duet Web Control 3.3.0

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122
=== Diagnostics ===
RepRapFirmware for Duet 2 WiFi/Ethernet version 3.3 (2021-06-15 21:44:54) running on Duet WiFi 1.02 or later + DueX5
Board ID: 08DGM-956GU-DJ3SW-6JKD8-3SN6T-TUJRD
Used output buffers: 3 of 24 (24 max)
=== RTOS ===
Static ram: 23876
Dynamic ram: 78700 of which 248 recycled
Never used RAM 8736, free system stack 106 words
Tasks: NETWORK(ready,126.5%,233) HEAT(delaying,0.7%,330) Move(notifyWait,2.7%,307) DUEX(notifyWait,0.0%,24) MAIN(running,794.8%,441) IDLE(ready,1.6%,29), total 926.3%
Owned mutexes: WiFi(NETWORK)
=== Platform ===
Last reset 01:25:35 ago, cause: software
Last software reset at 2021-07-06 10:02, reason: User, GCodes spinning, available RAM 9504, slot 0
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x0041f000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
Error status: 0x0c
Aux0 errors 0,0,0
Step timer max interval 0
MCU temperature: min 37.9, current 44.7, max 45.6
Supply voltage: min 23.8, current 23.9, max 24.3, under voltage events: 0, over voltage events: 0, power good: yes
Heap OK, handles allocated/used 99/4, heap memory allocated/used/recyclable 2048/1004/914, gc cycles 18
Driver 0: position 32220, ok, SG min/max 0/1023
Driver 1: position 27143, ok, SG min/max 0/1023
Driver 2: position 22475, ok, SG min/max 0/1023
Driver 3: position 0, ok, SG min/max 0/1023
Driver 4: position 0, ok, SG min/max 0/1023
Driver 5: position 0, standstill, SG min/max not available
Driver 6: position 0, standstill, SG min/max not available
[... 221 lines cut ...]
		M292 					; Hide the message
		M24 					; resume print
		
		M581 P0 S1 T2 R1 		; enable filament runout trigger
		
		break					; break the loop
	elif sensors.filamentMonitors[0].status == "ok"
		echo "filament Status ok"
		break					; break the loop due to false alarm
```

### What happened?

Expected result
Trigger executed immediately after filament sensor detected a ranout

Observed result
Trigger only execute after the resume print command is sent over DWC/PanelDue

Steps to reproduce
1. Print over SD
2. Cut filament to provoke run out alarm
3. DWC report Extruder[0] report no filament
4. wait for trigger to execute (does not execute)
5. press resume print (trigger execute at this moment)

x0rtrunks, 2021-07-08:
Please create a forum thread to discuss or provide a link to an existing discussion. Thanks.
https://forum.duet3d.com/

natthapolvanasrivilai, 2021-10-16:
Close the issue due to the changes in the implementation in 3.2+

https://duet3d.dozuki.com/Wiki/Connecting_and_configuring_filament-out_sensors#Section_RepRapFirmware_3_2_and_later