Duet3D/RepRapFirmware #543: [FeatureRequest]: Add command saving all current [running] config + vars into file

chernogorsky, 2021-10-03:
### Is your feature request related to a problem? Please describe.

Usually than I fine tune some complex designs (like for example tool changer/wiper) I made plenty of minor tweaks over  couple of days, and later move all that into gcodes/macros. For example: parking position (for me its global vars) or fine tuned steps for motors

But some times (and more than I liked to) I have to use emergency stop / power off to prevent some damage, so I lost a lot of fine tunes

### Describe the solution you propose.

If it's possible to add some command (or modify an existed one, like m500 but with parameters) just to dump all the config, including global vars in gcode format it would be just perfect 
or dump all the object state (it also has most of the config) into the  local duet file

### Describe alternatives you've considered

As a workaround I may try using self made script like m500.1.g with saving all the params by meown, but it would require me adding all the intersted parts manually and would require handling on every new params I want to play with

### Provide any additional context or information.

Ex. Photos, mockups, etc.

T3P3, 2023-05-04:
related to this FR: [Variables and parameter reporting](https://github.com/Duet3D/RepRapFirmware/issues/823)

@chernogorsky you can access all the global variables from the Object model so you could echo those to a file.

Dumping the whole, running, config to a file is a separate issue as that requires accessing elements that are not (yet) in the object model.

T3P3, 2026-08-24:
These issues cover some OM changes (not all) that would be needed:
#701 #759 #760 #700

T3P3, 2026-09-16:
Forum thread: https://forum.duet3d.com/topic/36005