Duet3D/RepRapFirmware #549: [FeatureRequest]: Add G15 G16 Polar coordinate

raffa1799, 2021-11-04:
### Is your feature request related to a problem? Please describe.

Ex. I am always frustrated when I write Gcode by hand with only a technical drawing and I need to make the cutter move in some points that I cannot calculate or that I cannot calculate and that I can pass thanks to the polar coordinates.

### Describe the solution you propose.

Ex. It would be great if... 
G16                  ; (polar coordinates on)
G1 X8 Y0 R0.1 Z-0.163 F3.0           ;X for leight and Y for angle        
G1 X8 Y60.0                                                                                      
G1 X8 Y120.0
G1 X8 Y180.0
G1 X8 Y240.0
G1 X8 Y300.0
 G15              ; (polar coordinates off)

### Describe alternatives you've considered

Ex. A workaround exists but it is tedius, for example...

### Provide any additional context or information.

Ex. Photos, mockups, etc.

dc42, 2022-03-29:
Please start a thread on the forum at https://forum.duet3d.com/ so that we can gauge how much interest there is in this feature.

T3P3, 2023-05-04:
@raffa1799 I see no discussion about this request on the gcode forum, I am closing this as not planned for now. We would consider this in the future if there is still a requirement for it, please open a discussion on the forum if that's the case.

>